OpenCPN/OpenCPN #4574: libs/serial/CMakelists.txt catkin tests statement leads to build failures

antonim24, 2025-06-05:
**Describe the bug**
The if statement in `libs/serial/CMakelists.txt` on line 27 will always evaluate to true. The syntax here incorrect for CMake, and also hard to understand the intent since it is a double negative. This is not a problem if you already have the package installed, but by default it should not be entering that if statement and thus requiring the package.

This is a nuisance for new developers as it is not required and therefore should not cause the build to fail.

Offending line:
```cmake
if (NOT ${CATKIN_ENABLE_TESTING} MATCHES "OFF")
```

Fix:
```cmake
if (CATKIN_ENABLE_TESTING)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Verify the catkin package is not installed.
2. Try configuring CMake.
3. See that there is a build failure as it cannot find the catkin package.

**Expected behavior**
Expect the if-statement to not get executed as by default `CATKIN_ENABLE_TESTING` is OFF. The build should configure and build as expected.

**Screenshots**
CMake build warning/error output:
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Ubuntu
 - Version 22.04.5

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: N/A
 - Andorid version: N/A

**Additional context**
N/A.

leamas, 2025-06-05:
I see your pain. However, on ubuntu you should have libcxx-serial-dev installed, and this should that libserial isn't built at all.

That is not to say that this bug is invalid, on the contrary. Just that it should be possible to work around on Ubuntu.

antonim24, 2025-06-06:
Ok I have done a lot of looking around and digging into it. Here is what I did and found:

1. I did a full clean of all the build dependencies off my system, removed all OpenCPN related items etc.
2. Fresh clone of OpenCPN. Followed the steps above exactly.
3. Catkin is still not installed with libcxx-serial-dev. This seems to have been a manual step and you can see it in libs/serial/Makefile.
4. Had no problems building this time and didn't run into the error. Turns out I had catkin_enable_testing on at some point, so the CMake cache had some extra items added. Even if it is turned on again, you have to delete the cache and start fresh otherwise it will try doing some stuff with catkin.
5. By default, the steps listed in the developer will result in a build failure. You have to build with "-DOCPN_DISTRO_BUILD=ON" to avoid a problem with "wx/uilocale.h" in one of the tests. The minimum version required of that file is 3.1.6, but the version that got installed was 3.0.5. Not sure if that is something that should be accounted for in the control file.
6. OCPN_DISTRO_BUILD is never defined as an option anywhere. Seems like you only pass it in via the command line and then a few places read the variable.

If you want to close out this issue that is fine. This CMakeLists.txt itself works, even with the wonky syntax in there.

leamas, 2025-06-08:
As sorted out elsewher this is basically about cmake version on Focal being too old. The fix is to install the python wheel distributed by the cmake project. The file *ci/generic-build-debian.sh* contains hints how to do this.

Closiing

closed 2025-06-08 by leamas