OpenCPN/OpenCPN #4577: Dashboard: Add dampening factor for True Wind Direction?

davidsanner, 2025-06-06:
Wind speed & angle have a configurable dampening factor.  

Could the angle dampening factor also be use to dampen "True Wind Direction" as it currently updates so frequently it's difficult to read?

Looks like the calculation is based on true wind and heading true.
So that's ground wind in degrees true?  Option for direction magnetic?

Hakansv, 2025-06-08:
As far as I can find, the wind damping filter only affects instruments for apparent and true wind angle and speed. There may not be a filter for the wind direction instrument or history chart.
The wind direction handle is quit complex and need to be explored more.  
Of course, it is best if there are opportunities to dampen the source and the data sequence for wind you describe seems completely unnecessary. But I assume you've looked for settings to control the sensor itself?

davidsanner, 2025-06-17:
I have a B&G WS310 (N2K) and you can't adjust the update rate, smoothing, etc. (They assume you'll use their MFD and adjust there.)

I like the fast update rate for the wind display / graphical arrow, it's just about as quick as looking up to my masthead windex w/ less neck strain ;)      On my N2k Furuno wind display the arrow smoothing is adjustable but again I have it fast.  The digital wind speed on the display I believe always stays at 1Hz though probably goes through a smoothing function.

Digital display of wind direction (and maybe heading) ideal setup might be a smoothing function with some sort of limit on update rate - goldilocks values?

I found this for B&G:  H5000 display has a 4Hz update rate for sailing data, including wind information.  I don't know if my eyes/brain would keep up but they may have done some testing before they picked that update rate?