OpenCPN/OpenCPN #4582: Linux Build Test Action is Failing on master

antonim24, 2025-06-09:
**Describe the bug**
The Linux build tests are failing in the master branch and the ci configuration incorrectly marked it as passing. This can be seen on the latest master branch build/test job for linux: https://github.com/OpenCPN/OpenCPN/actions/runs/15521814271/job/43696013914

See screenshots below

**To Reproduce**
Steps to reproduce the behavior:
1. Click on Actions tab in github
2. In the left navigation pane, click on "Linux build"
3. On the right side of the screen, set the branch to "master". You can also see this on any of the recent builds.
4. Click on the most recent build.
5. Select any of the two Ubuntu builds.
6. Open the "Test" job and scroll to the bottom. You will see one test failed, and one test resulted in a program abort.

**Expected behavior**
That when some of the tests fail, the CI job fails. And the tests should be passing.

**Screenshots**
Failed Tests:
[image]

Output from GTest for the failed tests:
[image]

**Desktop (please complete the following information if applicable):**
 - OS: N/A
 - Version N/A

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: N/A
 - Andorid version: N/A

**Additional context**
N/A

Crosswind, 2025-08-04:
I can see a third test failing which was introduced by 3058d601ebb825c98c9ca72cc9d2773770a0b907: `DriverRegistry.RegisterDriver` as part of the `n2k_tests.cpp`.

The test against the Swedish locale has an easy fix. The GitHub runners don't ship the Swedish locale by default. The patch for this would be:

In `ci/github-pre-build.sh` in the Linux setup.

```bash
	        # Unit tests require US and Swedish locales
	        sudo apt-get -qq install language-pack-us language-pack-sv
```

I am also installing the US language pack because it is also tested for the US locale. IMHO this should be included as well for a consistent environment setup.

Also, the workflow should be changed to fail on failing tests. Otherwise, this situation is likely to get worse with more failing tests staying unnoticed. This should be done after the current failures have been fixed. To keep the information on failing tests, I would change the workflow step to:

```yaml
      - name: Test
        working-directory: build
        run: CTEST_OUTPUT_ON_FAILURE=1 make run-tests
```

I am not submitting a PR because I have no idea how to fix the other two broken tests. @bdbcat may have the best idea on the `DriverRegistry.RegisterDriver` test. I couldn't trace back the `IpcServer.Commands` failure becaues the workflow logs don't go back that far.

In case I should still submit a PR for my change proposals, let me know.

leamas, 2025-08-04:
Indeed. See also #4077

antonim24, 2026-02-15:
@leamas Yeah I think it is good to close

leamas, 2026-02-15:
Ok. Let's do that then. Thanks for reporting!

closed 2026-02-15 by leamas