OpenCPN/OpenCPN #4583: Dutch Inland ENC charts bug: charts disappear when zoomed out or panning

Arne2024, 2025-06-09:
**Describe the bug**
Dutch inland ENC charts disappear when zooming out or when panning along the zoomed in charts.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'OpenCPN 5.10.2'
2. Click on chart downloader, download ''Dutch Inland ENC Charts'', do same for Belgian ENC charts, can both be build as ENC charts in OpenCPN
3. Zoom any part of the Dutch charts and then zoom out (larger scale) or move / pan between charts
4. Dutch inland ENC charts disappear, the Belgian inland ENC remain visible.

**Expected behavior**
when zooming ENC charts details and base charts should remain visible. Now they disappear seemingly at random when zoom out, or when panning from one chart to other Dutch ENC charts.

**Screenshots**
Added screenshots to help explain problem.

**Desktop (please complete the following information if applicable):**
 - OS:  macOS
 - Version Sequoia 15.5
 - Apple M1 MacBook 

[image]
[image]

Hakansv, 2025-06-09:
I'm on W10. 
These charts does sometimes have unusual scales compared to other and no overview files are present so it seems they have disappeared when zooming out.  It's normally not a problem when navigating since the scale use to be big and there's time for "preparation". The zoomed out quick panning is another thing.
Some things I use to test:
Try to "Prepare all ENC Charts" for the charts in question. 
Increase (temporary?) the "Vector chart Zoom/Scale factor" (Shortcut "D" can be useful)
Especially if other chart types are present one sometimes need to click a chart bar rectangle to "call them"

RooieDirk, 2025-07-13:
The Dutch (free) ENC suck. At least a little.;-) 
There are only detail scale charts, so zooming out means you have a lot of cells on screen. All of this cell need to be calculated and stitch together what might cause over asking the computing power. 
Playing with the Zoom/Scale factor does alter the moment the chart are disappear, setting the Options-> VectorChartDetail->Reduced detail at small scale and the Additional detail reduction at small scale, will help to reduce the computing workload.