OpenCPN/OpenCPN #4584: ENC chart: Bitmap Wreck Symbol in Grey (and depth) unreadable

Arne2024, 2025-06-09:
**Describe the bug**
The Wreck Symbols are overwritten by ''WK'' text and unreadable, do not zoom properly and too light

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Charts.'
2. Click on 'Download Inland ENC Belgium or Netherlands'
3. Scroll down to any are with WK 
4. See error and refer to screenshot

**Expected behavior**
The Wreck symbol should be shown on ENC chart, preferably just next text "WK" and the symbols should scale when zooming in, and possibly change colour to dark type.

**Screenshots**
see attached

**Desktop (please complete the following information if applicable):**
 - OS: macOS Sequoia 
 - Version 15.5
 - MacBook Pro M1 processor, 2020
 - OpenCPN 5.10.2
 - Inland ENC for Belgium, Netherlands

I have seen other reports open same issue in past but could not find back the issue or resolution.

[image]

[image]

bdbcat, 2025-06-11:
Corrected text position.
The symbology of the WRECK itself follows S52 recommendations.  The color is light grey, since the clearance depth over the wreck is deeper than the safety depth.  The symbol is not meant to scale on zoom-in, same behavior as all S52 symbols.

Arne2024, 2025-07-22:
this can be closed. 
Thank you for fixing this in the new version 5.12.
It now shows Wreck text and bitmap with depth side by side.

closed 2025-07-22 by Arne2024