OpenCPN/OpenCPN #4587: Route props: ETA and ETD are not updated when depature is changed.

Hakansv, 2025-06-11:
**Describe the bug**
If departure date is changed in route props only the first "start" row for ETA is updated. Not the subsequent ETA rows and neither of the ETD rows.

**To Reproduce**
Steps to reproduce the behavior:
1. Show a route properties.
2. Change departure date or time.
3. The first ETA, "start" row is updated but not the rest of the route.
4. See shot.

**Expected behavior**
All date and time for the route would be updated when departure is changed. (As it does for a speed change)

**Desktop (please complete the following information if applicable):**
 - OS: [Win10]
 - OCPN Version [Git master including: 3fa6c1248 ]

**Screenshots**
[image]

Hakansv, 2025-06-11:
To complete:
The above is valid for an old route.
For a newly created route all ETA/ETD rows are nicely updated when departure is changed.

bdbcat, 2025-06-11:
This problem results from an error on original navobj.xml import to navobj.db.
Correction is understood, coming soon.
Beta testers may need to export and re-import older routes if this problem is encountered in real life.
Production users will have no problem.