OpenCPN/OpenCPN #4589: ContextMenuCallback not invoked for plugins using API 1.20

TwoCanPlugIn, 2025-06-11:
If a plugin implements API 1.20, perhaps to use the Notification infrastructure (Eg.RaiseNotification API), existing Context Menu Items registered using AddCanvasContextMenuItem do not have the corresponding callback function OnContextMenuItemCallback invoked.

This breaks existing plugins and requires them to implement OnContextMenuItemCallbackExt instead,

The problem appears to be in canvasmenu.cpp in

```
if ((version_major * 100) + version_minor >= 120) {
.....
  opencpn_plugin_120 *ppi =
      dynamic_cast<opencpn_plugin_120 *>(pimis->m_pplugin);
  if (ppi)
    ppi->OnContextMenuItemCallbackExt(target_id, object_ident,
                                      object_type, zlat, zlon);

} else {
  pimis->m_pplugin->OnContextMenuItemCallback(pimis->id);
}
```
Is this intended ?

antipole2, 2025-06-16:
@TwoCanPlugin - please reopen this one.

When a context menu creates a subMenu with a list of items, the callback gives the id of the subMenu item.

As a result of this current issue,  a canvasContextMenu callback uses the old `OnContextMenuItemCallback()` callback
whereas a waypoint, route, track or AIS context menu uses the new `OnContextMenuItemCallbackExt()`

But... if the canvasContextMenu is set to display a submenu list, as supported in the extended version, then `OnContextMenuItemCallback()` is passed the id of the menuItem owning the submenu list and not the subMenu item.

So this change has broken subMenu lists for canvas context menus.

antipole2, 2025-06-16:
I suggest a better solution would be for the original call to `OnContextMenuItemCallbackExt()` to be restored (as before this issue) and for canvas context menu callbacks, to also call the `OnContextMenuItemCallback()`.
That way, legacy plugins would get what they are used to and new plugins would get the extended information (like position).

TwoCanPlugIn, 2025-06-17:
Re-opening as requested.

My brief testing didn't extend to sub-menus, so sorry I didn't notice that was also failing.

antipole2, 2025-06-18:
By my testing, the new 'Ext' APIa work as they should again.
I also reverted to JavaScript_pi v2.1.1 which uses the non-ext calls and that worked as it did.
So all good, I believe.

Thanks, Dave.

If @TwoCanPlugIn happy, we can close this?

TwoCanPlugIn, 2025-06-18:
My test plugin (aka Racing plugin) supports both AddCanvasContextMenuItem and AddCanvasContextMenuItemExt, and these both appear to work as expected in the most recent build. Note that it doesn't create any SubMenus so I can't verify if that works. 

I'm happy to close if Tony is happy that sub menus are working as expected.

antipole2, 2025-06-18:
Submenus working fine.  Can close.