OpenCPN/OpenCPN #4590: Cannot use two serial connections in parallel

leamas, 2025-06-12:
**Describe the bug**
With two serial connections, both of which fine when tested separately,  things does not work when both are enabled.

**To Reproduce**

1. Running on a Orange Pi 5, basically a beefed up Rpi 5 with RK3588 and an NVME port (running on a SSD)
2. Connect ttyS1 to internal GPS board which sends N0183 GPS position data at 9600 bps 
3. Connect a Navicom RT 750 AIS VHF using a USB dongle at ttyUSB0. This sends AIS and position data from an internal GPS at 4800 bps
4. Set up both ports as input connections.
5. Note that while the VHF has a list of AIS targets these are not visible in OpenCPN

**Expected behavior**
The AIS targets should be visible

**Desktop (please complete the following information if applicable):**
 -  XFCE, Debian 12 (Orange pi 5 port)

**Smartphone/Tablet (please complete the following information if applicable):**
- Orange pi 5

**Additional context**
If I disable the GPS data on ttyS1  the AIS targets from ttyUSB0 becomes visible.

This is not about general load, opencpn is at 4-5% CPU usage when this happens.

Same results on flatpak/stable, flatpak/beta and native build from (almost) current master.

Scary one, this.

bdbcat, 2025-06-12:
Is the TTYS1 connection rigged as a real or virual USB interface?
Trying to parse whether this is a USB problem, or a serial lib problem.
Can you run two instances of minicom, or picocom, and see both data streams without OCPN involved?

leamas, 2025-06-12:
ttyS1 is (by definition) a real device, not USB involved.

leamas, 2025-06-12:
> Can you run two instances of minicom, or picocom, and see both data streams without OCPN involved?

hm... I can only reproduce this on my yacht, writing this at home. Will make this test tomorrow.

Hakansv, 2025-06-12:
A thought: AIS data @4800 is normally not enough. As of standard and "normal" setup 38400 is needed. Does the extensive AIS data choke the port buffer? I'd never tested AIS data on that slow baude rate.

bdbcat, 2025-06-12:
Normally, AIS requires 38400, even with not many targets.
Do I understand that you see AIS targets from ttyUSB0 if the TTYS1 is disconnected?

bdbcat, 2025-06-12:
I ask because I am setting up a test now with two USB serial ports, both 4800, connected to GNSS receivers.  No results yet, still configuring udev...

leamas, 2025-06-12:
> Do I understand that you see AIS targets from ttyUSB0 if the TTYS1 is disconnected?

Yes

bdbcat, 2025-06-12:
Tested with two USB serial connections, both 4800 GNSS source.
Working fine, either one, or both  together.
Note sentence sources alternating in picture.

[image]

leamas, 2025-06-13:
> Note sentence sources alternating in picture.

https://opencpn-manuals.github.io/main/opencpn-dev/messaging-debug.html#_timestamp_1

leamas, 2025-06-13:
Message from S/y Equipage:

> Can you run two instances of minicom, or picocom, and see both data streams without OCPN involved?

Yes

leamas, 2025-06-18:
hm... After fetching soldering iron I now have a new configuration:

  - One input port at 38400 bps for AIS data from the VHF
  - One output port for at 4800 bps  for autopilot and instruments.

The interesting fact is that this actually seems to work, while my previous shared port for input/output did not. I still suspect there is some bug lurking around here, probably related to having input and output for different devices on the same port. 

But since I cannot reproduce this any more without hardware changes: closing

closed 2025-06-18 by leamas