OpenCPN/OpenCPN #4617: Extra dashboard 'instrument' Speed and course history

pa2wlt, 2025-06-21:
While sailing solo and sleeping in blocks (or just when sailing and only looking at the instruments every now and then), I like to have insight in the trend data for SOG and COG instead of just the actual values, so that I have better insight in what is going on.

**context**
Dashboard already has an excellent 'instrument' that does this for wind: **Windhistory**. Only few people seem to be aware of this. It shows TWS and TWD over time. Not just the values read, but also a trendline. That is very handy to have on your screen especially in instable weather. I will add a hyperlink to an image here of that wind history.

**Request**
Now, I would like to have a '**SOG and COG history**' instrument as well, so that I can see what these values where during the last hour: is the SOG increasing or decreasing, did COG change? These things are not trivial, especially not while sailing solo on long ocean passages.

This **SOG and COG history" can use the same layout as the Windhistory that is already available. For the Y-axis on the right hand side, we might use numerical values: 000 - 030 - ... 330, since that makes more sense for COG.

rgleason, 2025-06-21:
Ok, now hit OK. And then Enable it  and from the Toolbar Menu
Click on the Plots graphs icon.  Then you will see the main menu and be able to select which graphs to show.

pa2wlt, 2025-06-22:
Thx Rick. Thx for the instructions! I was completely unaware of Plots.

Seems very powerful. If it would have been in the dashboard, I would have found and used it way earlier.

Now ... Seans Youtube video shows there should be an X and an Y-axis. On my screen (Mac) I don't see either. Also not after restarting OpenCPN. Without info on the axis, the plots are not very useful yet. I'll see if it the axis show on Win11.

rgleason, 2025-06-22:
I haven't tried Plots for quite some time. I was able to simulate nmea0183 data stream with VDRplayer and get the graphs working in Plots.  when you right click in the Plots field a menu comes up with "Configuration " at the bottom. I notice that transparency does not work well now, and there needs to be some work on improving graphics and line thicknesses.  But it does show the data you are looking for, but I don't believe it supports nmea2000 data.

[image]

[image]

pa2wlt, 2025-06-22:
thx @rgleason ! I installed it on the win11 machine and there it had an x- and y-axis. Cool! Visually a few things could be improved, indeed. Though this is very handy already! Thx again. Shall I close this feature request then?