OpenCPN/OpenCPN #4621: Grib_pi incorrect display in 2K resolution

bc38, 2025-06-22:
- The position projection data line has moved down.
- The slider is very, very short
- The icon resolution is pixelated, but it is an SVG vector
The position projection needs an option to enable and disable display. Not many people understand what this is for, and it takes up a lot of space.

[image]

[image]

bdbcat, 2025-06-23:
What is the exact screen resolution tested here?
What system/OS?

bc38, 2025-06-24:
> What is the exact screen resolution tested here? What system/OS?

Windows 11
2880x1920, Scale 250%

bdbcat, 2025-06-25:
Screenshot, please?