OpenCPN/OpenCPN #4635: AIS track recording: Track disappears when stopping the recording

LeaseOnLife, 2025-06-28:
**Describe the bug**
when performing "Record track", then "stop recording" for a moving AIS target, the track disappears, and any new track is also not shown

**To Reproduce**
query a moving AIS target (which is painting a magenta colored track) and select "Record Track". The track color changes to a green tint. Then select "Stop tracking" and the complete track of that vessel disappears from the screen. While the target continues to move, no more track is shown.

**Expected behavior**
I think "Stop tracking" should revert the tracking into it's original state, before "record track". so the track should show colored in magenta, and the max length of it should be as defined in "Settings/Ships/AIS targets".

The section that has been recorded should also be displayed, possibly in a different color.

**Desktop (please complete the following information if applicable):**
 - OS: Debian Trixie on PI and amd64

LeaseOnLife, 2025-06-30:
> thanks for your diagnosis. What I can add is, it worked before (to described expectations), for example in 5.8.4 (i just tested this). So somehwere in the 5.10 series this functionality changed to something undesired. I may play with some 5.10 versions a bit more to find the time when this occured.

Update: 5.10.0 does show the problem already, just a bit different. When clicking "stop tracking" the old, recorded track stays visible on the screen in green (good), but the ship does not show the continuation of its track from the point the tracking was stopped (undesired!?) . A later change must be responsible to make the recorded track disappear from the screen when clicking "stop tracking".

transmitterdan, 2025-06-30:
I have submitted a pull-request that I think fixes things so they work as one might expect.

https://github.com/OpenCPN/OpenCPN/pull/4637

leamas, 2025-07-18:
#4637 is merged. 

@LeaseOnLife: could you please test this so we eventually can close this issue?

closed 2025-07-18 by LeaseOnLife