OpenCPN/OpenCPN #4640: Strange creation of monitor.log empty of data

Corsair-63, 2025-06-30:
**Describe the bug**
after some steps in the monitoring window recording the data it creates a monitor.log file with only the header but empty of data.

**To Reproduce**
it has been complicated to reproduce the creation of the file.

I've started with the option I noticed that the file was created empty of data. first of all, I erased all files in the [destination](https://www.cruisersforum.com/forums/tags/destination.html) folder.
opened ocpn
gone to [monitor](https://www.cruisersforum.com/forums/tags/monitor.html) window
exchange the saving path to desired one and chosen VDR
start recording, no monitor.log file created but proper monitor.csv file ?????
exchange to CSV with now the option of overwriting, done recording and again no monitor.log and proper monitor.csv file.
exchange to Default and now created both monitor.log and .csv both with all valid data.
exchange now to CSV and start recording and now properly recorded monitor.csv and erased all the content from monitor log.

monitor log after all these events
```
# timestamp_format: EPOCH_MILLIS
# Created at: Mon Jun 30 19:58:40 2025
 
received_at,protocol,msg_type,source,raw_data
```

**Expected behavior**
The creation of the file should be consistent with the option, and if should have to be created, it shouldn't be empty of data.

**Screenshots**

[image]

[image]

[image]

[image]

and Excel warning on opening monitor.csv after stop recording but opencpn opened.

[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11 ARM64 (24H2)
 - Version Opencpn 5.11.3

leamas, 2025-07-01:
Hm...

As I understand it, the header-only monitor.log is created when one selects this format. If the logging isn't started after this only the  header-only log  you describe is left.

But: is this a bug? I'm not convinced. Are you?

Corsair-63, 2025-07-01:
Hi Leamas,
I've reproduced easily.
what is clear, for me now, in that depending on options there are some files created:
- Default: creates a monitor.log file
- VDR: creates a monitor.csv file
- CSV: creates a monitor.csv file

now first if rewritten option is not thick why the files are overwritten (of any kind)

- recording folder clean of files
- making a recording with "Default" option creates a monitor.log file with all the info (attached)
- now switch to VDR, overwritten option not active, create a monitor.csv valid but monitor.log wiped for data, why?? supposed that this last file is not forming parte of this recording.
- switch again to the remaining option CSV, overwritten option not active, create a monitor.csv valid with new data but monitor.log wiped for data again.

from my point of view the monitor.csv files should not be rewritten with latest option, should have to be renumbered e.g. monitor1.csv, and so on, and the same for the .log files if are they eeded in "default" option.
and why is rewritten monitor.log when is not needed in CSV and VDR options?

[monitor.csv](https://github.com/user-attachments/files/21003947/monitor.csv)

[monitor.log](https://github.com/user-attachments/files/21003956/monitor.log)

[monitor1.log](https://github.com/user-attachments/files/21003960/monitor1.log)

Hakansv, 2025-07-01:
Win 10. Today's Git master
The selection as of attached shot creates a monitor.csv file, see an example. That file can not be replayed by the external plugin VDR even though it is called "VDR". 
The file created by the external plugin itself is the next example. That file can nicely be replayed by the VDR plugin.

monitor.csv
```
# timestamp_format: EPOCH_MILLIS
# Created at: Tue Jul  1 01:37:57 2025 
received_at,protocol,msg_type,source,raw_data
1751326687609,NMEA0183,TCP:signalk.stupan.se:10114,AIALL,"!AIVDM,1,1,,A,13@oQKG00KPT>?8Ptqe2rjD>0<8P,0*24<0D><0A>"
1751326687609,NMEA0183,COM6,AIVDM,"!AIVDM,1,1,,A,13@oQKG00KPT>?8Ptqe2rjD>0<8P,0*24<0D><0A>"
1751326687609,NMEA0183,TCP:signalk.stupan.se:10114,AIALL,"!AIVDM,1,1,,A,B3uFhfP000=SqU`?pKD03wSUoP06,0*2B<0D><0A>"
1751326687609,NMEA0183,COM6,AIVDM,"!AIVDM,1,1,,A,B3uFhfP000=SqU`?pKD03wSUoP06,0*2B<0D><0A>"
```

vdr_20250630T235439Z.csv
```
timestamp,type,id,message
2025-07-01T01:54:39.689Z,AIS,,"!AIVDM,1,1,,A,13fJn>WP00wEEfRMOBfN4?w@28GQ,0*54"
2025-07-01T01:54:40.467Z,AIS,,"!AIVDM,1,1,,A,13fJq9g000OEE?dMO>>LomcB00RB,0*7E"
2025-07-01T01:54:41.598Z,AIS,,"!AIVDM,1,1,,A,13@o@kG001Ph4fFPqPCnlmA>0`G@,0*0F"
```

[image]

leamas, 2025-07-02:
@Hakansv This issue is about the log files created. Let's stick to that, the vdr format is another thing. We could discuss this in zulip if required.

@Corsair-63 

>  recording folder clean of files
  - making a recording with "Default" option creates a monitor.log file with all the info (attached)
  - now switch to VDR, overwritten option not active, create a monitor.csv valid but monitor.log wiped for data, why? supposed that this last file is not forming parte of this recording.
  -  switch again to the remaining option CSV, overwritten option not active, create a monitor.csv valid with new data but monitor.log wiped for data again.

OK, sorry for being slow learner.  This is indeed a bug. I have made a patch which makes sure that  the existing monitor.log isn't wiped out when switching to for example VDR.

rgleason, 2025-07-02:
This is because OpenCPN chose not to follow VDR lead with regard to timestamps.
It will now probably be impossible to convert the timestamps to be compatible with VDR, therefore the discussion was moved to Zulip. 

Everyone understood this would be a problem when the decision was made.

Corsair-63, 2025-08-03:
Tested with latest version 5.12.2, performed the following test:
overwriting not selected
right click and select Default and exchanged the path to desired warehouse folder, started recording in monitor.log, stopped recording and perfect file created.
right click and selected now VDR, automatically exchanged file name to monitor.csv, started recording after a while stopped, file created with correct data, also, monitor.log not swept of data and kept the previos recording data, perfect.
right click and now selected CSV, kept the file name as monmitor.csv, start recording ... stopped recording, and everything correct the monitor.csv overwritten with latest data.
..
gone again to right click and select Default, the name of the file doesn't exchange to monitor.log and keeps monitor.csv, start recording ... stopped recording and file monitor.csv created with latest data.

why when selected Default it doesn't exchange to monitor.log on the second and subsequents rounds?
or why not calling all the as only monitor.log either monitor.csv?

leamas, 2026-02-14:
@Corsair-63  I don't  really see a nug here, perhaps an RFE. Thoughts?

leamas, 2026-02-17:
> I consider, absolutely from my point of view, that the file extension should be fixed e.g. for:

No. The difference between the extensions is that  .csv files are valid csv files which can be loaded by  for example a spreadsheet. OTOH, the .log file  is not.

I'll  have a look on the empty files.