OpenCPN/OpenCPN #4643: Android - Language

Rasbats, 2025-07-03:
**Describe the bug**
During installation there is no option to choose the language used by OpenCPN. The installation uses the system language. Once installed 'User Interface' does not allow a change of language. OpenCPN does not appear in the list of apps where the language can be changed.

**To Reproduce**
Steps to reproduce the behavior:
1. Install (No option to change language)
2. Click on 'Options/Useer Inferface'
3. Scroll down to 'Language'
4. No option to choose any other language than the system language

**Expected behavior**
In Linux and Windows it is possible to choose the language on installation. And change it later if required.

**Desktop (please complete the following information if applicable):**
 - OS: Android
 - Version 13

**Tablet**
 - Device: Lenovo M10
 - Andorid version: 13

**Additional context**
Because of this bug it is not possible to change the font for 'CurrentValue' for languages other than English. 
**GetOCPNScaledFont_PlugIn()** cannot be used for the frcurrents_pi plugin in languages other than English.

bcn2, 2025-07-04:
As far as I remember in the case of Android the language is system wide - to change the language for an application change the system language. No chance to play within one app.

Rasbats, 2025-07-07:
bcn2... thanks, that explains the inability to change language for Android OpenCPN

For frcurrents_pi the problem is probably that I did not translate.  `GetOCPNScaledFont_PlugIn("CurrentValue", 16); ` should have been `GetOCPNScaledFont_PlugIn(_("CurrentValue"), 16); `.
Need to update my .pot file.

@bdbcat : Think I understand what was going wrong.

closed 2025-07-07 by Rasbats