OpenCPN/OpenCPN #4649: incorrect console position, linux : raspbian bookworm , opencpn 5.10.2.2

voilier-martineke, 2025-07-07:
**Describe the bug**
When the console first appears, it is centered on the screen and not on the right like in other operating systems.
To put it back in place, calling the options screen and closing it without doing anything returns the console to the correct location.
Another solution is to resize the OpenCPN window.

**To Reproduce**
use a navigate to, to bring up the console.

**Expected behavior**

**Screenshots**

[image]
[image]
[image]

**Desktop :**
 - OS: linux , RPI 4 b
 - Version : raspian bookworm (the last)

**Additional context**
in portrait or landscape mode, same behavior

voilier-martineke, 2025-07-09:
Hi
Yes we are using wayland.
Thanks

bdbcat, 2025-12-04:
Corrected for upcoming OpenCPN 5.14.
Closing

closed 2025-12-04 by bdbcat