OpenCPN/OpenCPN #4651: Give an information if a GPS source switching has taken place on the signalk side.

aviatorhh, 2025-07-11:
**Is your feature request related to a problem? Please describe.**
I have multiple sources that are managed by a signalk server which itself handles fallbacks. I did not recognize that my primary GPS source had failed and fell back to the GPS source from AIS.

**Describe the solution you'd like**
A information display (alert maybe) if the GPS source has switched on the SK side not within O. As fas as I know that information is transported (src ID).

**Describe alternatives you've considered**
Not yet but maybe use Dashboard_SK to display some info on the source.

RooieDirk, 2025-07-13:
As far as I know this is already done in the current beta. At least I see regular a notification about changing gps source on my test system.

aviatorhh, 2025-07-13:
Can you confirm the source change listed in the connection list of your settings or the source within signalk?

RooieDirk, 2025-07-14:
In my setup I have only one gps connected to a rpi running the signalK server. This gps is  multi system and if the fix is shifting from GPS to Galileo (In nmea $GLRMC to $GPRMC), I get a notification in OpenCPN .
Notification are also logged by OpenCPN, and looks like this:

06/15/2025  08:33:58 CEST
GNSS position fix priority shift:   
 -> SignalK 192.168.10.20:3000

aviatorhh, 2025-07-14:
Ok, that sounds correct. I will check on my side.

aviatorhh, 2025-07-14:
I have on my SK side a GPS UDP channel receiving NMEA0183 and an AIS serial channel getting AIS data incl. AIVDO. This then is fed into O via SK.
When I turn of the UDP/NMEA data feed on the SK side, I can see a change in O but only seeing the GPS signal indicator changing from a signal strength view to the green dot/globe view (because AIS does not send satellite data). No other notifications comes up. When switching the GPS back on again, the indicator goes back to normal without any notification either.
All tested on O 5.11.1 (d2bf89fe06fa64c4acdd548ffd73daa596f0c0a6)

aviatorhh, 2025-07-14:
[image]