OpenCPN/OpenCPN #4652: Duplicate tabs in Options->Charts

TeaM-TL, 2025-07-14:
**Describe the bug**
Duplicate tabs (o-charts and Chart Downloader) in Options->Charts

**Screenshots**
I attach screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: macOS
 - Version sequia 15.5

[image]

DuboisPa, 2025-07-16:
same on Windows 10. 
plugin desactivated and reactivated corrects the problem

nocheinpeter, 2025-07-21:
Same on Windows 11, opencpn 5.12.0.

Bug appears not direct but after a while, maybe then editing some settings.

Hakansv, 2026-01-06:
On W11.
Using code for upcoming 5.14, i.e. Git master 5.13, I can't reproduce this.

leamas, 2026-02-16:
@TeaM-TL  and others: Is anyone able to reproduce this on 5.13 beta?

leamas, 2026-07-27:
Ping?