OpenCPN/OpenCPN #4654: Clear the plugin "black list" - .\opencpn\load_stamps - when OCPN reinstall or upgrade.

Hakansv, 2025-07-19:
**Is your feature request related to a problem? Please describe.**
Background: 
A user experienced that the internal Dashboard plugin was no longer loading, nor was it available in Options->Plugins. 
The reason was probably that something had happened on the computer at one point when OCPN was started. Possibly a voltage drop that interfered with the computer just as the Dashboard was loading, resulting in the creation of a "load_stamp" entry. This record then remained even when OCPN was reinstalled and the user had no future access to Dashboard_pi. 

**Describe the solution you'd like**
That the solution to the problem was to manually clear "load_stamp" from the Dashboard is not something we can expect a user to know. However, we can expect the user to try reinstalling to possibly resolve the issue. 

It is reasonable that the "load_stamp" is cleared during new/re-install for a new attempt to install the current plugin.

leamas, 2025-07-20:
>  I suppose a program file like Dashboard_pi.dll can never been "corrupted" by OCPN since it is never written to

It is actually installed read-only on all OSes, soi it cannot be written to.  

True also on Windows as long as we don't run with administrator privileges.  Which basically is not supported.

leamas, 2025-07-21:
> It is a sensitive and fragile solution to create a file that O will delete if everything goes w

On the contrary: It is a robust solution which does exactly what it should: reflects that a load attempt is started and eventually completed. Since a failed load in many cases leads to a crash it is not really possible to use some logic inside of OpenCPN instead.

> Also, if the plugin is the cause for a crash there is no changes for a user to update to a possible new edition of that plugin since it's not even listed in Options->Plugins as long the file is present in "load_stamps". Am I correct here?

No. If starting in "Safe mode" the plugins could be installed, uninstalled, etc even if they are not loaded

With the fixes here nothing worse than that user would need to reinstall a plugin should happen. I think this is a "good enough" solution

Hakansv, 2025-07-21:
> With the fixes here nothing worse than that user would need to reinstall a plugin should happen. I think this is a "good enough" solution

Yes, as long the user accept to use "Safe mode" the very first time after a crash. If not the plugin in question will not be shown in Options->Plugins and no plugin update is possible.

leamas, 2025-07-21:
What you are looking for is just not possible. If a "bad" plugin is loaded (no safe-mode) opencpn crashes, and nothing more could be done. End of story.

This means that safe mode the only way to handle failed plugin loads. We just have to live with this fact.

leamas, 2025-07-21:
Safe mode is the correct solution, there is no other sane way to handle the situation when there is an unloadable plugin.

The perceived one-off usage is a windows thing where users are not used to using command line options. Could be handled by installing yet another icon like "OpenCPN (safe mode)" or so.

Safe mode is actually a well designed working way to handle plugin crashes. We should stick to it.