OpenCPN/OpenCPN #4656: COM connection does not transmit on v5.12.0

pelakuti, 2025-07-20:
**Describe the bug**
I am using the Raymarine autopilot plugin that generates a $NMEA sentence to be sent to a COM port. If I take a look to the debugging monitor window, the sentence is generated but not send to the COM. It is in red colour and filtered as "VIRTUAL".

**Desktop (please complete the following information if applicable):**
 - OS: win11
 - ocpn 5.12
  -plugin: lastest version, downloaded from opencpn app.

**Additional context**
The developer of the plugin says in previous versions this worked fine. The sentences generated by the plugin was sent to a COM port.

I upload a LOG.

Link to plugin develover's github where I oppened an issue: https://github.com/BerndCirotzki/raymarine_autopilot_pi/issues/37

[monitor.log](https://github.com/user-attachments/files/21337920/monitor.log)

bdbcat, 2025-07-21:
I did find an error in OpenCPN v5.12.0.  NMEA0183 message coming from plugins were accidentally dropped.  This affects several plugins.  I fixed the error, and you may like to verify by installing v5.12.0 from the daily Cloudsmith builds, when complete.

BerndCirotzki, 2025-07-22:
Hello Dave,
I took a look what you have commited to fix this .... I'm a little confused ... So I have a question if it rigth what you did.
This is what you did : 

if (!(n0183_msg->source->iface != "virtual)")) {
      return;

... But could it ever be true ?     "virtual)" could this be a typo ? .. And double "not" ( ! and != ) ... is equal, isn't it ... ?
I tested myself with that : (works also)

if (n0183_msg->source->iface != "virtual") {
      return;

So could it be that you disabled that what you actually wanted to do with the check of the "source_driver" ?
This is only a question, nut more.

Bernd

pelakuti, 2025-07-22:
I wait till the answer for this last question, and I close the issue! I just want to say thanks to Bernd and Dave. Amazing job!

closed 2025-07-22 by pelakuti

pelakuti, 2025-11-17:
Hi Dave,
I updated the other day to the lastest version of OpenCPN and the issue of the "virtual" is back. Are you aware of this? 
Thank you for your amazig effort and great job!

leamas, 2025-11-17:
If you have an immediate problem you could just apply that patch and rebuild. Otherwise you could just sit back and wait, we should be able to merge this in a not that distant future. 

It's currently waiting for some more testing ~but~ by the test dept a k a @Hakansv

pelakuti, 2025-11-17:
> If you have an immediate problem you could just apply that patch and rebuild. Otherwise you could just sit back and wait, we should be able to merge this in a not that distant future.
> 
> It's currently waiting for some more testing ~but~ by the test dept a k a [@Hakansv](https://github.com/Hakansv)

Thanks once again!
Absolutely any problem for waiting. I would do by myself but sadly I do not have the skills :(

Thank you!

Hakansv, 2026-01-06:
Late respons but I have now tested to send N0183 from a plugin, "AIS Voyage Data", to OCPN's multiplexer and it's sent out by any output connection, Serial (COM) as well as Network UDP. Also visible in Data Monitor.

leamas, 2026-02-07:
@pelakuti  OK to close?

pelakuti, 2026-02-07:
Can I try one more time onboard to be sure? I would like to test it one more time, and if it does not work I will upload here the log.