OpenCPN/OpenCPN #4659: Send RMC messages without active route

leamas, 2025-07-22:
**Is your feature request related to a problem? Please describe.**
My clipper NMEA repeater does not display speed and heading without an active route

**Describe the solution you'd like**
RMC (position, heading speed) should always be sent, active route or not. There is not actual reason to limit this to when there is only an active toute

**Describe alternatives you've considered**
None

**Additional context**
THis happens because RMC is sent as part of UpdateAutopilotN0183 in autopilot_output.cpp. This is obviously  only done when there is an active route

bdbcat, 2025-07-22:
leamas...
Please report your configuration for connections.

TwoCanPlugIn, 2025-07-23:
Dave,

You should have read my post in full. I'm not complaining about the bug resolved by 0e468a0.

I'm complaining that this enhancement aims to address either an end user configuration error or a limitation of a third party product and will have unintended consequences.

To keep it short and simple. For users with a NMEA 0183 GPS, unless they configure a filter, it will double the transmission rate of RMC sentences. For TwoCan users with a NMEA 2000 GPS receiver they will similarly have to configure a filter and there is the risk of a data loop.

TwoCanPlugIn, 2025-07-24:
My observations, both on 5.12.0 and on build c00b5a using virtual com ports and udp are that it seems to work albeit with some weird behaviour that warants further investigation.

This is what I feed OpenCPN with (COM5, Input)

```
$GPGGA,000412.29,3824.32,S,14435.44,E,1,10,0.98,1113.0,M,-21.3,M,25,10*5F
$GPGLL,3824.32,S,14435.44,E ,000412.29,A,A*53
$GPRMC,000412.29,A,3824.32,S,14435.44,E,015.60,231.00,240725,8.5,E,A*26
$WIMWV,296.2,R,007.5,N,A*2E
$SDDPT,006.6,-0.3,100.0*7A
$HCHDG,228.0,003.0,E,0.0,W*5B
$SDVHW,,T,,M,012.0,N,006.5,K*42
$IIRSA,01.54,A,,V*79
$ERRPM,E,0,2000,,A*42
$ERXDR,P,300000,P,ENGINEOIL#0,C,89.5,C,ENGINE#0,U,13.4,V,ALTERNATOR#0,G,250,,Engine#0*47
$ERXDR,E,15,P,Fuel#0,E,80,P,FRESHWATER#0#0,E,50,P,FRESHWATER#1,E,75,P,FRESHWATER#2*67
```

This is what I get back, when a waypoint is activated. No difference between UDP or Serial output

```
$GPGGA,000136.23,3824.15,S,14435.74,E,1,10,0.98,1113.0,M,-21.3,M,25,10*50
$GPGLL,3824.15,S,14435.74,E ,000136.23,A,A*5C
$GPRMC,000136.23,A,3824.15,S,14435.74,E,007.20,233.00,240725,8.5,E,A*2C
$WIMWV,288.0,R,007.4,N,A*22
$SDDPT,017.6,-0.3,100.0*7A
$HCHDG,230.0,003.0,E,0.0,W*52
$SDVHW,,T,,M,012.0,N,006.5,K*42
$IIRSA,-06.54,A,,V*53
$ECRMB,A,0.007,L,001,002,3837.305,S,14415.804,E,20.442,229.859,7.199,V,A*58
$ECRMC,000136.23,A,3824.150,S,14435.740,E,7.200,231.000,240725,8.500,E,A*3F
$ECAPB,A,A,0.007,L,N,V,V,229.879,T,002,229.859,T,229.859,T*39
$ECXTE,A,A,0.007,L,N*48
$ERRPM,E,0,2000,,A*42
$ERXDR,P,300000,P,ENGINEOIL#0,C,89.5,C,ENGINE#0,U,13.4,V,ALTERNATOR#0,G,250,,Engine#0*47
$ERXDR,E,15,P,Fuel#0,E,80,P,FRESHWATER#0#0,E,50,P,FRESHWATER#1,E,75,P,FRESHWATER#2*67
```

OpenCPN is creating a RMC sentence. I don't recall it ever doing that. Should it be ? Possibly not if there is a valid source of RMC sentences such as GPS and those sentences, unless filtered are also forwarded. And there must be a source of RMC for OpenCPN to obtain SOG & COG values.

Also as Tony has observed, I see a lot of "red" entries in data monitor. Does that mean they are being dropped ?

[image]

TwoCanPlugIn, 2025-07-24:
If some users require a RMC sentence for the A/P then fine, but kind of strange that as the industry is  moving away from NMEA 0183 towards NMEA 2000 that you haven't had this request before. 

If users have needed it, how has this requirement been met in the past? The NMEA Converter plugin ?

If OpenCPN is navigating and there is a source of RMC sentences in the original stream, then I'm not sure whether sending another RMC sentence is a good idea, especially if it is a slow speed 4800 baud NMEA 0183 connection. Either a configuration setting to enable/disable it or guidance to filtering. 

I don't have enough experience with other autopilots to know whether any require RMC sentences . It makes little sense to me, because there is no information in a RMC sentence about the next waypoint (no course to steer or cross track error).

Worth noting that if RMC is not available in the general stream then they're also not getting SOG or COG.

If the autopilot has some rudimentary MFD capabilities (stores and allows the user to select a waypoint or create a route, then I could understand that it would need a source of navigation data, either RMC, GLL or GGA sentence in order to calculate a course to steer or cross track error. but I don't have any knowledge of these.

For example my old Robertson . It is currently configured to use XTE mode and uses either APB, RMB or XTE sentenes. Eg. anything containing cross track error and it just steers to maintain cross track error at zero. The other mode, bearing mode (set internally with jumpers), uses APB, RMB, BWR, BOD (essentially anything that has a magnetic bearing to the destination).

bdbcat, 2025-07-24:
Looking at git history, I find that we have been sending RMC since at least the 2020 time frame.
So evidently it does no harm, and might be useful to some.
It stays.

TwoCanPlugIn, 2025-07-24:
Yes, you are correct about the history.

I'm trying to clean up the last few bits in TwoCan, namely autopilot stuff, converting XTE, RMB, APB & ActiveLegInfo into PGN's 129283, 129284 & 129285.

I noticed the sentences emitted by OpenCPN lack the FAA mode indicator field which then prompted me to remember #3452.

At the moment I can't faithfully generate PGN 129283 because the XTE sentence emitted by OpenCPN lacks this field, so I resort to an assumption. 

I know there was a historical reason for OpenCPN not to support this (older autopilots etc), but moving forward I think there should be a configuration option to enable OpenCPN to correctly support the NMEA 183 v4.x standard where most of these sentences (APB, RMB, RMC, XTE etc.) do have a FAA Mode indicator field.

leamas, 2025-07-25:
Back to original issue: my Clipper NMEA repeater does not display speed/heading without an active route. @bdbcat said: 

> Please report your configuration for connections.

[image]

   - /dev/ttyS! is an internal connection to a waveshare GPS module sending  positions, speed, quality, etc.
   - /dev/vhf is a navicom vhf which sends AIS data
   - /dev/instrument is a waveshare combined RS485 and RS232 interface. The 485 data goes to a Raymarine ST2000 auto tiller pilot, the 232 goes to a Clipper  GPS repeater which shows the things I need to sail on open waters: Speed, heading, distance and bearing to waypoint.

Overall, this is a nice setup which needs very little current . It runs on an orange pi which basically is an enhanced RPi 5 with more  CPU power, an nvme interface and requiring less current than the RPi 5.

However, the clipper repeater is problematic. Long story short it requires RMC/RMB in pairs so to speak. If I for example pass the GNRMC from the waveshare GPS is starts complaining about missing RMB  even if ECRMB is available -- it wants a GNRMB to match  the GNRMC. The GPS doesn't know anything about routes so this will not happen.

Arguably, this is a clipper bug. However, since it's otherwise nice I  want to keep it.

But after digging more into this just sending  RMC  when there is no active route will not make the trick, the repeater then also needs dummy RMB to be usable..  Which boils down to a device specific option  "Send dummy Clipper GPS data" or so which indeed  sends ECRMB and ECRMC also when there is no route.  I have made a quick hack without GUI and it seems to work.

Such an option would not affect the general behaviour and would thus don't have the risks involved in always sending RMC.

bdbcat, 2025-07-25:
IIUC, the situation at the moment is:
Using latest master,
1.  RMC is being sent correctly to  /dev/instruments.
2.  You need to synthesize RMB somewhere, to match the RMC normally available, and send to /dev/instruments.

This tickles my memory, somehow, about the unusual requirements of the Clipper.

leamas, 2025-07-25:
Yes.  And to handle this we need a specific "handle broken clipper" option rather than changing the general behaviour.

I have it partly working when there is an active route by filtering, only passing ECRMB and ECRMC (thanks, Håkan).  This is using current master.

I have a dirty patch which sends ECRMB (faked) and ECRMC also when there is no active route. This is verified to work. Will eventually make a PR with GUI and persistence based on this.

bdbcat, 2025-07-25:
Probably will suggest that the option NOT be added to the GUI, bust require manual config file edit.
Which of course brings up the question of "Where are all these non-GUI options in config file documented?"

I resist the notion of adding GUI options to correct third party devices, by name/brand.  It is a slippery slope...

antipole2, 2025-08-04:
This sort of issue is easily handled by a simple JavaScript to push out the required sentences.