OpenCPN/OpenCPN #4670: Teleported positions

antipole2, 2025-07-25:
**Describe the bug**
I have had an another weird passage where our ship is frequently teleported to some distant place and back.  This is always accompanied by an AIS proximity warning.  See the yellow track.

[image]

To try and get round another problem with loss of position (will report in separate issue) I had increased the Watch Dog timeout to 30 seconds.
Examining the notifications, you can see  there were 17 position priority shifts to use AIVDO.

[image]

The problem is that it was getting the position of some other AIS target - not us.  Each of the teleports was to an AIS target.

My priority settings are 

[image]

I have my MMSI number entered in the settings Own Ship.
In the MMSI Properties tab I have my own MMSI set to ‘ignore’

It seems that when OCPN falls back on AIVDO, it is using the position from some random target rather than own ship.

**To Reproduce**
Steps to reproduce the behavior:
1. Interrupt the normal receipt of a fix so falls back onto !AIVDO
2. Or raise priority of AVDO

**Expected behavior**
It should use own ship's position, not some other ship!

**Desktop (please complete the following information if applicable):**
 - OS: MacOS 15.5
 - OCPN v5.12.0

TwoCanPlugIn, 2025-07-25:
Curious, what brand/model of AIS? 
Is it connected via NMEA 2000?
Can you capture the NMEA2000 log from your Actisense device?

antipole2, 2025-07-26:
My setup:
Garmin AIS600 -> N2K -> Actisense W2K-1 -> NMEA0183 -> OpenCPN

I cannot read the N2K-1 log as I do not have Windows.

I have just decoded a !AIVDO sentence and it is not from my (UK) MMSI but a Netherlands MMSI (I am in the Netherlands).

I suspect the W2K-1 and will raise this with Actisense.

bdbcat, 2025-07-26:
Please leave this issue up.
I propose to implement (3), or something like it, for OCPN v5.14.