OpenCPN/OpenCPN #4671: W2K-1 de-blocking issue?

antipole2, 2025-07-26:
**Describe the bug**
I have an Actisense W2K-1 converting my N2K data to NMEA0183, which is fed to OpenCPN.

The flow of NMEA0183 data, as observed in the Data Monitor is irregular.  It usually arrives in bursts - 10-12 messages with the same timestamp, each block arriving about a second or two apart.  But sometimes there is a pause in the data - 4-5 seconds is common but sometimes it can be 30 seconds or more, when OCPN reports position lost and reopens the channel.

@nohal has pointed out ([see here](https://github.com/OpenCPN/OpenCPN/issues/4670)) that the W2K-1 is buffering up the messages and sending them as a block.  Hence 10-12 messages with the same timestamp from OCPN.
He says a message can be split across different blocks.

OCPN must be de-blocking the messages and, maybe, reassembling split messages.

I suspect there is an issue here - perhaps an edge condition - whereby OCPN loses its place in the process.

**To Reproduce**

1. You need an W2K-1 feeding data as NMEA0183.
2. You need to configure the W2K-1 to not process PGNs 129025 & 127250.  These rapid update PGNs swamp OCPN with 10 messages /second.
3. In DataMonitor filter to observe just the input from the channel.
4. You will see the data arriving in n blocks about every second.

***The issue: From time to time there is a pause in the data.***

You can enable Watch Dog and watch the flow of, say, GLL.  Most of the time, it updates every second or two.
Sometimes it takes 5-6 seconds and sometimes much longer.  These delays correspond to the pauses in the flow of processed data.

I have logged Monitor data available.

Toggling the OCPN connection off and back on always restarts the flow, so I suspect OCPN rather than the W2K-1

My W2K-1 is on firmware v 1.398.3436

**Expected behavior**
The data flow should be regular.

**Desktop (please complete the following information if applicable):**
 - OS: MacOS 
 - Version 15.5

**Additional context**
OpenCPN v5.12.0

leamas, 2025-07-26:
> @nohal has pointed out (https://github.com/OpenCPN/OpenCPN/issues/4670) that the W2K-1 is buffering up the messages and sending them as a block. Hence 10-12 messages with the same timestamp from OCPN.

This is not only about the Actiisense. See: https://opencpn-manuals.github.io/main/opencpn-dev/messaging-debug.html#_timestamp_1

TwoCanPlugIn, 2025-07-26:
Worth reading [this](https://actisense.com/news/how-does-nmea-0183-to-nmea-2000-data-conversion-work/?srsltid=AfmBOorbb8JQ8suVkNvk9DlUq7nO3s25pO2U-m4E96vbsFchGvkoMF2R)  to understand how Actisense devices receive NMEA 2000, buffer, convert & transmit NMEA 0183. It explains why you get bursts of traffic.

Really dumb question, which also applies to #4670 why not connect OpenCPN to the W2K-1 using NMEA 2000 instead of NMEA 0183?

DuboisPa, 2025-07-26:
> Worth reading [this](https://actisense.com/news/how-does-nmea-0183-to-nmea-2000-data-conversion-work/?srsltid=AfmBOorbb8JQ8suVkNvk9DlUq7nO3s25pO2U-m4E96vbsFchGvkoMF2R) to understand how Actisense [...]

because using wifi to multiple devices ?

bdbcat, 2025-07-26:
re:
"You need to configure the W2K-1 to not process PGNs 129025 & 127250. These rapid update PGNs swamp OCPN with 10 messages /second."

Quick test on Windows, with only the one connection from the W2K-1, not a particularly fast machine, shows no problem processing the 10 Hz stream.

bdbcat, 2025-07-26:
The above test is with the Windows PC directly connected to the W2K-1  Wifi access point.  No intervening router.
IIRC, you have your W2K-1 connected by Wfif through some other local router?  If so, please share details of that.

bdbcat, 2025-07-26:
Exact same test on Mac, direct connect to W2K-1, same results.  No problem with 10 Hz.  No dropouts.

TwoCanPlugIn, 2025-07-26:
Reading Pavel's observations that the individual NMEA sentences are concatenated and sent in a block of 1490 bytes in length makes me think that Actisense have arbitrarily chosen the Ethernet/WiFi MTU size, so they can send one packet and not worry about fragmentation.

I just performed a quick & dirty test on Windows, concatenating a number of sentences into one chunk and sending via UDP and it seemed to be OK. But I didn't fragment any sentences over the chunks so probably warrants a deeper examination.

Also wondering if Tony is running any plugins, perhaps his own and there is something weird in SetNMEASentence?

And of course any network related stuff that I'm not I'm not in a position to comment on.

@DuboisPa 

> because using wifi to multiple devices ?
The W2K-1 has three servers that can be configured for any of the conversion protocols.

antipole2, 2025-07-26:
I am seeing the input flow stop every 1-3 minutes.
It recovers after the Watch Dog time out provided I am not also receiving N2K.

However, I have done test with a direct connection to the W2K-1 (not via my router) and that seems to be fine.
I had one hang after an hour (with both 0183 and n2k inputs), which is not ideal but a vast improvment.

I conclude that my rather basic router is the problem.  I intend to replace it over winter with a modern one.

So I propose to close this issue.  I will reopen if I encounter the problem after upgrading my router.

antipole2, 2025-07-26:
> Worth reading [this](https://actisense.com/news/how-does-nmea-0183-to-nmea-2000-data-conversion-work/?srsltid=AfmBOorbb8JQ8suVkNvk9DlUq7nO3s25pO2U-m4E96vbsFchGvkoMF2R) to understand how Actisense [...]

Thanks for informative links.

The reason I am using NMEA0183 input is that I have an iPad at the wheel station and other iOS devices running TZ iBoat.  It is an excellent app with reasonable charts and a very good display of AIS point of nearest approach. (I will not go into all the other good stuff about it here.) It is NMEA0183-based.  If I only have N2K inputs, TZ iBoat does not get fed and does not display AIS targets and CPAs.  It would be a lot of work to decode theN2K stuff and convert to NMEA0183, which the W2K-1 does for me.

TwoCanPlugIn, 2025-07-27:
@antipole2 

> The reason I am using NMEA0183 input is that I have an iPad at the wheel station and other iOS devices running TZ iBoat.

Configure one of the other two W2K-1 servers as NMEA 2000. Your other devices will still get NMEA0183 from the  currently configured server and configure OpenCPN to get its input from the NMEA 2000 server. 

I would also expect you teleporting problem to disappear in that configuration.