OpenCPN/OpenCPN #4674: Excessive logging in "Production" code

soggygeek, 2025-07-27:
**Describe the bug**
In 48 hours of almost doing nothing, OpenCPN generated a 16MB log file.  Most of it is repetitive screen monitoring stuff.

**To Reproduce**
Steps to reproduce the behavior:
1. OpenCPN 5.12 on Windows 11
2. 1 NMEA 183 input (combined GPS and AIS on one USB port), limited to 1/sec
3. UDP out to 192.168.8.1 (all) and Marine Traffic (AIS sentences only)
4. NO OTHER ACTIVITY
5. Let it run for 48 hours.  

Around every 6 seconds, about 10-12 messages are logged, same stuff over and over and over.  This looks like debug that should have been removed 

11:42:36.665 MESSAGE chcanv.cpp:2428 Metrics:  m_display_size_mm: 478     g_Platform->getDisplaySize():  1920:1080   
11:42:36.665 MESSAGE chcanv.cpp:2434 monitor size: 1920 1080
11:42:36.884 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
11:42:36.898 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
11:42:36.948 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
11:42:36.952 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
11:42:37.234 MESSAGE chcanv.cpp:2428 Metrics:  m_display_size_mm: 478     g_Platform->getDisplaySize():  1920:1080   
11:42:37.234 MESSAGE chcanv.cpp:2434 monitor size: 1920 1080
11:42:43.066 MESSAGE chcanv.cpp:2428 Metrics:  m_display_size_mm: 478     g_Platform->getDisplaySize():  1920:1080   
11:42:43.066 MESSAGE chcanv.cpp:2434 monitor size: 1920 1080
11:42:43.192 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
11:42:43.199 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
11:42:43.263 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
11:42:43.272 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
11:42:43.579 MESSAGE chcanv.cpp:2428 Metrics:  m_display_size_mm: 478     g_Platform->getDisplaySize():  1920:1080   
11:42:43.579 MESSAGE chcanv.cpp:2434 monitor size: 1920 1080
11:42:55.726 MESSAGE chcanv.cpp:2428 Metrics:  m_display_size_mm: 478     g_Platform->getDisplaySize():  1920:1080   
11:42:55.726 MESSAGE chcanv.cpp:2434 monitor size: 1920 1080
11:42:55.891 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
11:42:55.892 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
11:42:56.074 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
11:42:56.075 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
11:42:56.230 MESSAGE chcanv.cpp:2428 Metrics:  m_display_size_mm: 478     g_Platform->getDisplaySize():  1920:1080   
11:42:56.230 MESSAGE chcanv.cpp:2434 monitor size: 1920 1080
11:43:01.999 MESSAGE chcanv.cpp:2428 Metrics:  m_display_size_mm: 478     g_Platform->getDisplaySize():  1920:1080   
11:43:01.999 MESSAGE chcanv.cpp:2434 monitor size: 1920 1080
11:43:02.116 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
11:43:02.126 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
11:43:02.360 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
11:43:02.362 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
11:43:02.502 MESSAGE chcanv.cpp:2428 Metrics:  m_display_size_mm: 478     g_Platform->getDisplaySize():  1920:1080   
11:43:02.502 MESSAGE chcanv.cpp:2434 monitor size: 1920 1080

I have a full 48 hour log file saved, if you want to see it.

**Expected behavior**
At log level INFO, routine stuff like this should be logged this frequently

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
Windows 11

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: [e.g. Galaxy S20]
 - Andorid version: [e.g. 10]

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

bdbcat, 2025-07-29:
Sherry..
Good to hear that "something" has changed.
Please keep me posted on your steps to bring it back up to the functionality you need.
Especially that last critical step, whatever it turns out to be.

soggygeek, 2025-07-29:
It's baaaaack!  So have disabled Com3 again.  Will let you know if that does it.

bdbcat, 2025-07-29:
OK, a good clue.  I can replicate that locally.  I'll be back...

soggygeek, 2025-07-30:
OK, downloading now.  Still trying to track down whether the serial port is the (only) issue.  the behavior is not identical every time, and I can't figure out what the variable is.
My friend who also mentioned a huge log (v4.8.2) file said it was probable that he has a serial port (or two or three) that are defined in OpenCPN but not operating, as he is still trying to recover from a lightning strike.  Whether that means the port exists in Device Manager and just isn't connected to anything (that's working) or whether it's like mine where I have a port defined in OpenCPN, but the USB cable isn't plugged in, so there is no port there in Windows.
I will install your version and observe.

bdbcat, 2025-07-30:
I'll be watching.
Turns out the code for opening the "not there" serial port was executing an exception (throw()) every 2-3 seconds.  Still not clear how this impacted the size calculation in OCPN, which was leading to the to log flood.
But this sort of hi frequency exception is not a good thing, in any case.  So I removed it.

soggygeek, 2025-07-30:
I don't think that has solved the problem.
It seems to take awhile before it starts logging this repetitively:
```
11:43:02.116 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
11:43:02.126 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
11:43:02.360 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
11:43:02.362 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
```
(note, this is a copy of the message from a previous log file, I neglected to copy one from the boat computer today).
But it does eventually start logging.  I am not sure what kicks it off.  I was messing about on the computer quite a bit today--never letting it just run hands-off, so it was hard to draw definite conclusions.
Just before I left, I shut down the VNC Server, to make sure it is not that that is causing the issue, and restarted OpenCPN again with Com3 Enabled, but not defined on the system.  No plugins enabled. But with VNC down, I can't check it from home.  I may walk down after dinner to check on it, as I know you're anxious to try to track this down.

bdbcat, 2025-07-30:
OK, now I an suspicious of something in the display configuration.

soggygeek, 2025-07-31:
Well, I started it up at 6pm last night.  With Com3 Enabled and nothing on Com3 (not showing in Device Manager).  With RealVNC server shut down.  The startup sequence ended at 18:14:01

I forgot and left one plugin enabled, so after startup, I disabled the one plugin and again checked that Com3 was enabled. Looks like I finished that at 18:15:40.  And the logging was quiet for a whole 15 MINUTES.  I minimized OpenCPN and brought Chrome browser to the front (with our node-red monitoring system receiving a short wifi burst with data from our BMS every second, displaying the data on the screen, and doing a text line logged to disk every 10 seconds)

After 15 minutes of nothing in the log, we got this:
```
18:30:00.493 MESSAGE ocpn_frame.cpp:5754 LOGBOOK:  2025-07-30 10:30:00 UTC  GPS Lat    4.21026 Lon  100.60271 COG  180.34000 SOG   0.27 kts
18:34:29.525 MESSAGE chcanv.cpp:2429 Metrics:  m_display_size_mm: 478     g_Platform->getDisplaySize():  1920:1080   
```
And from then on, oodles of logging about display size and framebuffer

Screen is dark, no one is aboard, no mouse activity, nothing going on but the AIS/GPS input on one NMEA Com Port

06:12:25.323 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3

This message started being logged at 06:12 in the morning, which happens to be EXACTLY (almost to the second) 12 hours after startup. Infrequent at first. Note: We are +8 UTC and the computer is running on local time, not UTC.  But it gets more frequent as time goes on.  Here's a snip from 6 hours later.

`12:05:43.754 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:45.003 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:45.009 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:45.011 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:45.117 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:45.119 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:45.120 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:45.646 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:45.648 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:45.650 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:45.693 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:45.694 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:45.695 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:45.897 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:46.905 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:48.063 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:49.063 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:50.069 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:51.498 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:51.501 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:51.502 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:51.647 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:51.649 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:51.650 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:51.772 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:51.773 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:51.778 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:51.780 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:51.782 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:51.871 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:51.879 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:51.902 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:52.262 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:53.264 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:54.269 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:55.274 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:56.279 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:57.290 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:57.932 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:57.943 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:57.992 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:58.003 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:58.005 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:58.042 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:58.048 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:58.053 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:58.975 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:58.978 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:58.979 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:59.000 MESSAGE glChartCanvas.cpp:602 BuildFBO 3
12:05:59.001 MESSAGE glChartCanvas.cpp:1056 OpenGL-> Framebuffer OK, size = 1918
12:05:59.002 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
12:05:59.196 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 3
[... 86 lines cut ...]
13:08:09.980 MESSAGE chartdb.cpp:246 Purging unused chart from cache: D:\Charts\Valhalla\MalaysiaRev28\Pangkor.mbtiles
13:08:09.994 MESSAGE chartdb.cpp:1032 All chart in cache locked, size: 2
[... cut ...]
13:08:13.128 MESSAGE chcanv.cpp:2429 Metrics:  m_display_size_mm: 478     g_Platform->getDisplaySize():  1920:1080   
13:08:13.128 MESSAGE chcanv.cpp:2435 monitor size: 1920 1080
`
Attached is the full 8MB log file, zipped.  I made a couple of notes in the file, search for XXXX
Note that I realized after I shut down, that I still have 2 CM93 chart data files in my chart definition.  I don't know if that could have anything to do with the excessive logging.  I have removed one of the CM93 chart sets.

[opencpn.save.log.zip](https://github.com/user-attachments/files/21523451/opencpn.save.log.zip)

bdbcat, 2025-07-31:
Some analysis.
1.  Some process in the system is causing a nearly periodic resize/refresh cycle to OpenCPN.  Cannot reproduce locally, W10 or W11.  Appears not coming from inside OCPN, but rather injected from the OS.  I can only speculate that this effect is coming from some other app or service process running on the system.  Maybe the Chrome browser, which seems to be the only other GUI app running at the time?
2. The failure to reload a chart is due to memory exhaustion in the OCPN process.  Performed experiments to induce the memory problem by intentionally injecting an OnSize() event every second.  
Results:  On one system, (W10) the memory usage grows slowly, but steadily.  Using VS tools to track the source, the culprit seems to be the Intel OpenGL driver.  On another W11 system, memory use appears to be steady.  No persistent growth in memory footprint.
3.  As another test, I ran both systems with OpenGL disabled.  Result: very steady memory footprint, for hours.

tl;dr;
Nest step would be to try with OCPN solo on the machine, no other apps present, OCPN maximized.

bdbcat, 2025-07-31:
Also:
Seems clear the problem is not directly related to the Com port setup.

soggygeek, 2025-08-01:
More inconclusive results.
The last change I made (2025-07-31 13:20 local) was to remove the 2nd CM93 chart set, and restart OpenCPN.  Could it be the 2 CM93 chart sets causing the problem? (seems unlikely) I got a minute to check on OpenCPN on the boat earlier this evening, and 18 hours of running, the log file is only 80 KB!  I stopped in briefly to grab the logfile, and was amazed. The only change I made was to remove the 2nd CM93 chart set (I think that was the only change).  (and restart).
However, the change I did before that was to shut down the RealVNC server, which didn't seem to make any difference. So when I grabbed the log file, I re-started the RealVNC server, and now, just checking from home a couple of hours later, I see the screen resizing logging again.
I did not restart OpenCPN this evening.  But minutes after I checked the log file (and restarted RealVNC server), the logging started again.
However, RealVNC server has been down for days before, with no change in the bad behavior.
I did just now shut down Chrome, and the screen resizing logging has stopped.
So, RealVNC server up, and with me logged on

soggygeek, 2025-08-02:
So here's a question... would leaving the system set up with "center on my boat", and the boat is moving a bit because of poor GPS coverage, cause this problem?  I notice on the every-half-hour logbook entry, there is a little movement logged.

10:00:00.240 MESSAGE ocpn_frame.cpp:5754 LOGBOOK:  2025-08-02 02:00:00 UTC  GPS Lat    4.21028 Lon  100.60266 COG  333.75000 SOG   0.17 kts

I don't see the screen moving, so there must be some kind of damping going on.  But definitely see the nose of the boat extend periodically.

bdbcat, 2025-08-02:
Unlikely, unless you are zoomed very far in.
However, if the received position is so very different that it implies a jump to a new chart when in auto-centering, I suppose it is possible to get a screen refresh.

bdbcat, 2025-11-15:
Sherry...
Any chance I could get AnyDesk remote access to this machine?
Thanks
Dave

bdbcat, 2025-11-16:
This is what we know:
1.  For some reason, this system is calling for a "resize" of the main OpenCPN window, roughly every 5 or 6 seconds.
2.  The "resize() event is not being generated from within OpenCPN, except in three cases:
     a.  Initial startup, several times.
     b.  When a user manually resizes the OpenCPN app frame.
     c.  In split screen mode, when the user drags the screen divider., thus resizing each canvas.
3.  AFAIK, no other systems in the wild experience this effect.  Or at least not reported.
 
I certainly could edit the source and produce less logging during the resize event.  Perhaps we should do that.

However, if it were my machine, I would like to know where these resize events are coming from.  They certainly are affecting overall OpenCPN performance, at the very least.

Recommendations:  Minimize the configuration ruthlessly.  Remove all plugins. Remove all extraneous system drivers and apps.  Remove all OCPN data connections.  Empty (default) OCPN config file.  No AIS data.  OCPN just sitting there, in normal maximized mode.  You get the idea....

soggygeek, 2025-12-15:
I'm still looking at this.  It seems crazy, but my current theory is that if I have several things going on on the computer, and I MINIMIZE OpenCPN before bringing forward another app (usually Chrome, which stays up 24x7), it will go for hours just logging the position info on the half hour.
But something ticks it into "resize" mode. And I haven't been able to reliably reproduce that situation.

<removed comment about "ignored SARTs" as a possibility... our AIS range is currently about 1/4 mile due to no AIS antenna hooked up, Ignored MMSI list is empty (due to upgrades in O?) and not getting any SART alarms>

bdbcat, 2025-12-15:
Sherry...
I have pushed and built a patched version that might affect this behavior.  Relates to OpenCPN response to alert conditions while minimized.  Maybe....
Good Luck.