OpenCPN/OpenCPN #4683: Uninstall error in Windows

denelon, 2025-07-30:
**Describe the bug**
Uninstall failure in Windows

Related to:
* https://github.com/microsoft/winget-cli/issues/5628

**To Reproduce**
Steps to reproduce the behavior:
1. winget install OpenCPN.OpenCPN --version 5.8.4
2. winget upgrade opencpn

**Expected behavior**
OpenCPN version 5.8.4 is uninstalled and OpenCPN version 5.10.2 is installed

**Screenshots**
Screenshots in related GitHub issue for WinGet

**Desktop (please complete the following information if applicable):**
 - OS: [Windows 11]

**Additional context**
The default upgrade behavior in WinGet via the manifest is "uninstallPrevious". That's where we're seeing the failure occur. I've included screenshots and more descriptive text in the related issue.

denelon, 2025-07-30:
> Shouldn't this be filled against https://github.com/microsoft/winget-pkgs/tree/master/manifests/o/OpenCPN/OpenCPN/ ? The winget package, about which I hear the first time ever, is not maintained by the OpenCPN team.

We'd love to have automatically published releases. The wingetcreate tool is used in these flows in most cases, and it makes publishing the latest version to Windows (WinGet) users seamless. I'm sure we could get someone to draft a PR if you'd be willing to accept one.

Examples:
* https://github.com/JanDeDobbeleer/oh-my-posh/blob/main/.github/workflows/winget.yml
* https://github.com/microsoft/PowerToys/blob/main/.github/workflows/package-submissions.yml

denelon, 2025-07-30:
Based on:

>If upgrading from a previous version of OpenCPN, there is no need to un-install the old version. Simply install the new version and it will upgrade OpenCPN, saving all your existing configuration and preferences.

I'll draft a PR to change the upgradeBehavior

denelon, 2025-07-30:
I took a look at the docs and there is a dead link to the "runtime components" on https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:quick_start_guide:installation:windows

WinGet does have the ability to install any dependencies if they are in the "same source" if they are required.

The content around "Choose where to get apps" may also not be necessary for distribution via WinGet on Windows 11. Part of the process of submitting packages to the WinGet Community Repository involves several scanning steps that will eliminate the "pop up" warning about the package coming from the Internet commonly displayed in the browser download scenario.

WinGet is available on [Windows 10 version 1809](https://en.wikipedia.org/wiki/Windows_10,_version_1809) and later as well as Windows Server 2025.

I didn't find any instructions for uninstalling OpenCPN, but when I tried to do that manually I got an error:

[image]

leamas, 2025-07-30:
> Expected behavior

>> OpenCPN version 5.8.4 is uninstalled and OpenCPN version 5.10.2 is installed

Simply stated, there is no such expectation from OpenCPN's point of view. 

The expectation is that the procedure described in the manual (download the installer and run it) works , and as far as I know it does. Thus this bug should be closed, leaving this to the winget maintainers at https://github.com/microsoft/winget-cli/issues/5628.

Our policy is that OP normally closes bugs, so please do.

denelon, 2025-07-31:
Do you want me to file a new one for the uninstall failure with the screenshot above?

I'm happy to remove the WinGet associated parts.

closed 2025-07-31 by denelon