OpenCPN/OpenCPN #4684: OpenCPN uninstall doesn't remove directory on Windows

denelon, 2025-07-31:
**Describe the bug**
The uninstall fails on Windows with an error message about not being able to delete the directory OpenCPN is installed in.

**To Reproduce**
Steps to reproduce the behavior:
1. Install OpenCPN
2. Uninstall OpenCPN

**Expected behavior**
The application is uninstalled.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: [Windows 11]

leamas, 2025-07-31:
> Since we introduced the use of "..\AppData\Local\opencpn" for plugin handling I think the NSIS wasn't updated to optionally delete also that. That would be feasible to do?

Possible, but IMHO a bad idea. The plugins  is a personal setup, might contain items obtained from other sources than than the official catalog. Furthermore, they are in general supposed to remain compatible not least on windows.

Bottom line: this is better left as is. This is actually about revisiting a **very** long discussion for Linux about the same issue.

whindsaks, 2025-09-14:
```
RmDir $InstDir\lib\cmake\glew
RmDir $InstDir\include\GL
```
Will only delete empty directories and should be safe to add to your installer.