OpenCPN/OpenCPN #4685: OpenCPN suddenly not receiving data from MiniPlex anymore

Crosswind, 2025-07-31:
**Describe the bug**
Today, out of nothing, OpenCPN doesn't receive any nmea0183 messages from our MiniPlex anymore. It worked a couple of of hours ago and stopped. No configuration was touched. We restarted OpenCPN because we weren't using it. I have been trying to figure this out for a good 4 hours now without luck. Hence, coming here for help.

The problem started appearing on `5.10.2`. I then saw that `5.12.0` was just released so I went ahead and updated. It didn't fix the problem (I wasn't expecting it would but you never know).

What's notable: Using the software from the MiniPlex I can see that the multiplexer is working just fine. I can see the stream of NMEA messages. OpenCPN doesn't see anything.

I can't seen anything interesting in the OpenCPN log. [opencpn.log](https://github.com/user-attachments/files/21529264/opencpn.log)

Also, I have AIS connected via a COM port as well. This works without any issue. I can see the NMEA messages going into OpenCPN.

Let me know if I can provide further information. Any help is appreciated!

**Things I tried**
- Removed the entire configuration of the COM port from OpenCPN
- Updated from 5.10.2 to 5.12.0
- Verified the USB connection to the multiplexer is working by using the vendor-supplied software
- Tried differend baud rates, normally we use 19200 for that connection

**Screenshots**
[image]
The status of the COM port is reported to be operational.

**Desktop (please complete the following information if applicable):**
 - OS: Windows 
 - Version: 11

leamas, 2025-07-31:
Your screenshot actually shows that Opencpn is receiving data on that port -- the checkmark is only present if there is data.

Have you tried the new Data Monitor (shortcut: the 'E' key)? It might give a hint about what's happening here.

Crosswind, 2025-07-31:
@leamas yes, I have tried using the new Data Monitor. It doesn't show any data being transferred:

[image]

These red symbols show up every time I open the Data Monitor window. Maybe that's worth a separate bug report - doesn't seem to be connected to NMEA directly. When selecting "Alle Daten" I see output - just no NMEA from the port in question.
With the AIS being connected I can see NMEA traffic in the Data Monitor.

Crosswind, 2025-08-01:
When using `All Data` I can see OpenCPN's internal messages as well as the messages from COM5 (AIS) and the outgoing messages to COM4 (MiniPlex). I can't spot anything obvious as to why it is failing to read from it. (Failing may be the wrong term here

This is a sample from the Data Monitor window. 
We can see:
- That it is receiving messages on COM5 (AIVDO, etc.)
- That it is creating messages because a route is active (RMB, RMC, XTE, APB)
- That it is sending messages to COM4
- That it is rejecting messages going to COM4

[image]

(We are in Courseulles-sur-Mer in France - for anyone looking up the GPS coords lol)

Meanwhile on the MiniPlex:

[image]

I also figured out how to run OpenCPN with LogLevel `debug` and `trace`. I couldn't spot any obvious errors but I also didn't try to read the entire log. Let me know if any of these log files in full could be helpful.

Hakansv, 2025-08-01:
From your Miniplex shot no position is sent.
There are some wind and compass data than would have been visible in the Data Monitor.

Since only ECxxx are sent to COM4 I suppose you have set a output filter. That's very good.
Could it be you accidentally also have set filter on the input data? "Input filtering":  "Accept only sentences" shall be empty.

Hakansv, 2025-08-01:
OCPN does use valid incoming messages from a serial port. No problem here or for others.
I've seen serial to USB converters failed. Just a hint.
What about using Putty (Your on Win?) to see what's coming in on COM4 by another program?
Or, if you use serial converters temporarily shift the one now to COM5 to COM4 instead?

leamas, 2025-08-02:
Looking at the driver code,  it basically handles an incoming message like

1. Check if it begins with '$' or '!'
2. If test succeeds, update the rx counter. This counter is what makes the connections tab display a ✓.
3. It checks if the message passes the input filter.
4. If input filter passes, it generates an input event  which should be visible in the Data Monitor.

Which boils down to that the input messages seems filtered. Things to check: properties of the connection in the connection tab, under  Advanced: how/if is filtering applied.?

Crosswind, 2025-08-02:
> What about using Putty (Your on Win?) to see what's coming in on COM4 by another program? Or, if you use serial converters temporarily shift the one now to COM5 to COM4 instead?

I wrote a small Python program that reads from the serial port and outputs to the command line.

```python
import serial

s = serial.Serial('COM4')

while True:
    res = s.readline()
    print(res)
```

Sample result:
```
b'\\s:MX32-1*41\\!AIVDO,1,1,1,,B39THi000?vSNw75vpoQ3wlUoP06,0*04\r\n'
b'\\s:MX32-2*42\\$IIMWV,037,R,11.07,N,A*20\r\n'
b'\\s:MX32-2*42\\$IIVDR,,,,,,N*0E\r\n'
b'\\s:MX32-5*45\\$GPGGA,140642.00,4935.22053,N,00115.78902,W,1,,,,M,,M,,*69\r\n'
b'\\s:MX32-1*41\\$GPRMC,140642.00,A,4935.22053,N,00115.78902,W,0.027,,020825,,,A*63\r\n'
b'\\s:MX32-1*41\\$GPVTG,,T,,M,0.027,N,0.050,K,A*23\r\n'
b'\\s:MX32-2*42\\$IIVHW,321,T,321,M,00.00,N,00.00,K*55\r\n'
b'\\s:MX32-1*41\\$GPGGA,140642.00,4935.22053,N,00115.78902,W,1,11,0.84,3.0,M,46.8,M,,*42\r\n'
b'\\s:MX32-2*42\\$IIVPW,,N,,*1F\r\n'
b'\\s:MX32-1*41\\$GPGSA,M,3,11,21,20,16,05,07,04,29,09,06,30,,1.53,0.84,1.28*09\r\n'
```

I'm not sure about serial converters. I simply have a USB plugged into my AIS and MiniPlex. On Windows I have the correct driver installed so the ports show up as COM ports and are usable.

> Things to check: properties of the connection in the connection tab, under Advanced: how/if is filtering applied.?

This looks fairly standard to me.

[image]

This is the general connections tab:

[image]

Hakansv, 2025-08-02:
Strange, really.
If you can read COM4 by any other program the converter would be fine.
One test I'd have made: Delete the COM4 connection and make a new. Start with only input and correct baud rate as used by the python script.  If that works include the output and its filter.

leamas, 2025-08-02:
And I understand it (my German is not what it should be) you seem to have an empty list of accepted messages.   Or?

Hakansv, 2025-08-02:
One thought.
A sentence receive as of the Python script:

> b'\\s:MX32-1*41\\!AIVDO,1,1,1,,B39THi000?vSNw75vpoQ3wlUoP06,0*04\r\n'

If the beginning "b'\\s:MX32-1*41\\" is nothing from the Python scrip but actually what's sent from the device it can be the problem.
It could be the "*41" is interpreted as the end of the message (before a checksum) and parsing jumps to next sentence before the actual NMEA message is read?
In any case that beginning is not according to the NMEA0183 standard.
Others will have to judge my code interpretation.

Crosswind, 2025-08-02:
You are completely right. I’m glad I posted the output here. Should’ve done that earlier.

There’s a setting on the MiniPlex that prepends these blocks. I know I messed around with it during my debugging and couldn’t make sense out of it. I apparently forgot to turn it off again. I have no idea how it got switched on before when the error started appearing. 

I quickly verified that the blocks are gone with my python script and that’s the case. I assume OpenCPN will be just fine again.

I started researching this and it turns out this tag includes the original input interface the message was received on. Neat because that allows for fun things.

This tag is part of the NMEA 0183 v4 spec. I believe this should at least be accepted and then ignored by OpenCPN. I’ve read that AIS may include a timestamp there as well. `\s:<tag>\` is for source and `\c:<time>\` seems to be for timestamps.

I found a similar issue: https://github.com/SignalK/signalk-parser-nmea0183/issues/69

I have not contributed to OpenCPN so far but I would be willing to at least write the feature request and perhaps challenge myself with some implementation work.

closed 2025-08-03 by Crosswind