OpenCPN/OpenCPN #4691: command ALT-C to open and close Data Monitor

DuboisPa, 2025-08-02:
Data Monitor can be opened by Menu ou by ALT-C 
When opened, ALT-C must also be used to close it.

version 5.12.x

Crosswind, 2025-08-03:
`e` can also be used to open Data Monitor. Why do we have two shortcuts?

Wouldn’t it be better to have just one and use standard `Esc` to close the Data Monitor again? This also works with other modals.

leamas, 2025-08-04:
@DuboisPa :

> Data Monitor can be opened by Menu ou by ALT-C
When opened, ALT-C must also be used to close it.

I actually fail to understand this. Could you please expand a little?

rgleason, 2025-08-08:
E toggle Data Monitor (reserved for future)

leamas, 2025-08-10:
Two different things: The hotkey (listed by the '?' key) and the menu entry. The hotkey is always available whether menus are enabled or not.

The basic question still stands: Why would it be important to close a visible window with a key instead of using buttons or window decorations?

Hakansv, 2025-08-10:
The shortcut hint for Data Monitor is not correct. Alt-C is for to show the menu itself.
The menu text should read: "Data Monitor     E"

leamas, 2025-08-10:
>  Might be better to stick to "E" only?

Yes

DuboisPa, 2025-08-11:
Hello, 

Because some users like using their keyboard.  :-)
A hotkey or hotkey-combination is faster than quitting the keyboard, taking the mouse and click on the X to exit. 
You can use a hotkey to open, you can use a hotkey to quit.
When you're using a word processor (or something else), CTRL-S to save is faster than "mouse-menu-file-save".
Ergonomy is not only  to be nice, but to be more efficient.

DuboisPa, 2025-08-11:
Why should a user learn a specific OpenCPN key to **open** Data Monitor ?

leamas, 2025-08-11:
> Why should a user learn a specific OpenCPN key to open Data Monitor ?

Because there is no generic wau to open a specific window like this -- there is no platform out there which has a documented shortcut which opens the Data Monitor for obvious reasons.

TwoCanPlugIn, 2025-08-11:
Bored during dogwatch....

For Windows users, nice that you are reinventing 30+ years of established UI design.

Start by removing the confusion between  shortcut keys and access keys.

Shortcut keys may be OS wide and available within all apps. Eg. Ctl-C copy, Ctl-V paste, Ctl-F4 close a tab or a windows in an app. The latter on Windows (and it's equivalent in other OS's) should be the shortcut key to close the Data Monitor window. Nothing more, nothing less. 

There may also be application specific shortcut keys, such as the currently debated Alt-C for the Data Monitor. As Alec points out, unlike cut, copy, paste there is no standard shortcut key for a "Data Monitor".

Access keys are the keys that complement a mouse click to invoke a menu item. And on Windows, that is visually the underlined letter of a menu item.
Eg. Alt-F invokes the top level File menu. Within that menu, O invokes the Open File menu option. The menu item <ins>O</ins>pen has the letter "O" underlined.

And it is possible to have both a shortcut key and an access key. Eg.select some text in an app, Alt-E for the Edit menu, and either the access key "t" or Ctl-X for "Cu<ins>t</ins>"

I can't understand the purpose of debating something that has been standardised for so long, other than to point out a lack of consistency in O's use of shortcut and access keys.

DuboisPa, 2025-09-20:
As I will never have ALT+C, also to close Data Monitor, this issue can be closed.

rgleason, 2025-09-20:
You should close it. You are the Original Poster (OP).  OP close their Issues.

closed 2025-09-21 by DuboisPa