OpenCPN/OpenCPN #4692: NMEA 0183 sentences discarded when tag is included (v4 spec)

Crosswind, 2025-08-03:
**Describe the bug**
In cases where the NMEA talker includes a tag as part of the sentence, OpenCPN silently discards the message. The tag is part of the NMEA v4 spec and allows metadata to be included as part of the message.

I am using a ShipModul MiniPlex-2USB which allows to set tags on the sentences to identify the original input channel the sentence was received on.

[image]

Sample data that include these tags:
```
\s:MX32-2*42\$IIVPW,,N,,*1F
\s:MX32-1*41\!AIVDM,1,1,,B,13IjIK?P00OpTr:LIfHtlgwn2<0H,0*71
\s:MX32-1*41\!AIVDO,1,1,2,,B39THi000?v9Ck76Q8sQ3wuUoP06,0*15
\s:MX32-2*42\$IIVWR,028,L,12.14,N,06.25,M,,*21
\s:MX32-2*42\$IIVWT,022,L,12.36,N,06.36,M,,*2F
\s:MX32-2*42\$IIHDT,274,T*3D
\s:MX32-2*42\$IIWCV,,N,*20
\s:MX32-2*42\$IIWPL,,,,,*67
\s:MX32-2*42\$IIXTE,A,A,,R,N*79
\s:MX32-2*42\$IIBWC,,,,,,,,,,,N,*18
\s:MX32-1*41\!AIVDO,1,1,3,,B39THi000?v9Ck76Q8wQ3wP5oP06,0*55
\s:MX32-2*42\$IIDBT,021.65,f,006.60,M,003.56,F*21
\s:MX32-1*41\!AIVDM,1,1,,B,13P=Qn@013woG?BLQpACWC9:05Qd,0*6D
\s:MX32-2*42\$IIHDM,275,M*3C
```

I don't own the v4 spec but the general format is:
`\<tag-type>:<value>*<checksum>\`
In my case, it is the `source` tag. `c` tag is for sentence timestamps AFAIK.
The checksum is calculated in the same way the checksum for the general message is calculated.

FYI: The format of the source tag for my multiplexer is: `<UID of multiplexer>-<input interface number>`

**To Reproduce**

1. Use MPXConfig to enable tags on NMEA sentences
2. Configure COM port in OpenCPN
3. No messages are received on the COM port. Sending messages is possible.
4. After disabling tags, messages are received in OpenCPN again.

I assume other multiplexers have similar functionality and instructions may be different.

**Expected behavior**

Whether or not tags are parsed and contents usable inside OpenCPN, they should not lead to messages being discarded. This bug report focuses on the fact that messages containing valid NMEA v4 tags should be accepted.

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Version: Tested with OpenCPN versions 5.10 and 5.12

**Additional context:**
The finding was made as part of #4685. More details can be found there but this issue should contain the relevant information. Based on @bdbcat's comment, v4 tags should be supported but something is wrong here.

transmitterdan, 2025-08-03:
On my Windows system with latest git/master this data is parsed and reported in the comm diagnostic window:

[image]
 
Also, GNSS position is updated correctly:

[image]

transmitterdan, 2025-08-03:
If it doesn't work using some other transport scheme, then the problem is most likely specific to the transport method specific code.  All the downstream code seems to be happy with this data.

transmitterdan, 2025-08-03:
This "hack" may resolve this issue but I cannot test it:

```
bool StdSerialIo::SetOutMsg(const wxString& msg) {
  wxString msg_copy = msg;
  int SOM = msg_copy.Find("$");
  if (!SOM) msg_copy.Find("!");
  if (!SOM)
    return false;
  msg_copy = msg_copy.Mid(SOM);
  if (msg_copy.size() < 6)
    return false;
  m_out_que.Put(msg_copy.ToStdString());
  return true;
}
```

leamas, 2025-08-03:
@transmitterdan wrote:
> This "hack" may resolve this issue but I cannot test it: [...]
o
Or, without reverting to wxString
```
bool StdSerialIo::SetOutMsg(const wxString& msg) {
  int start_pos = msg.find('$');
  if (start_pos == std::string::npos) start_pos =  msg.find('!');;
  if (start_pos == stf:.string::npos) return false;
  m_out_queue.Put(msg.substr(start_pos));
  return true;
}
```

But the basic logging problem needs IMHO to be dealt with first.

transmitterdan, 2025-08-03:
Agreed. Why do the low-level I/O classes try to form complete messages? Why don't they just pass up the raw NMEA data to a common point that knows how to find SOM/EOM and form complete messages regardless of how the bytes come from all the various devices.  Then they could be logged at one and only one place in the code.

That would mean all the serial, net, etc. device I/O specific code has to be oblivious to NMEA and just become byte and/or sentence pushers.

It's a big refactor though.

transmitterdan, 2025-08-03:
> Agree. Plus, the input validation shouldn’t be transport method specific, right? In the end, an NMEA sentence would look identical whether it is transferred via serial connection or TCP.
> 
> Would it hurt removing this check entirely? Do you remember the exact purpose why this was introduced in the first place?

Yes, it's because the next level up code assumes it will always receive a fully formed sentence (abeit with maybe bad checksums or garbled bytes). Ergo, each transport handler has to find Start of Message and End of Message so fully formed NMEA sentences can be sent up to the next level. It's a hold-over from when there were only one or two transport types and no serious logging capability. Now, it's causing code duplication and preventing effective log handling.

TwoCanPlugIn, 2025-08-03:
This is one from the memory banks.

The bug I filed on this aeons ago is probably lost in the detritus of flyspray.

IIRC OpenCPN has always discarded tags. The bug I filed was because OpenCPN was clearly discarding tags and wasn't forwarding the tags onto any output ports. I think at the time someone was interesting in adding timestamps.

Resurrecting the old code...

This is what is being sent:
```
\c:1754242770*39\$GPGGA,163929.85,3824.11,S,14435.82,E,1,10,0.98,1113.0,M,-21.3,M,25,10*53
\c:1754242770*39\$GPGLL,3824.11,S,14435.82,E ,163929.85,A,A*5F
\c:1754242770*39\$GPRMC,163929.85,A,3824.11,S,14435.82,E,016.80,235.00,030825,8.5,E,A*29
```
This is what is received:
```
$GPGGA,163932.95,3824.11,S,14435.81,E,1,10,0.98,1113.0,M,-21.3,M,25,10*5B
$GPGLL,3824.11,S,14435.81,E ,163932.95,A,A*57
$GPRMC,163932.95,A,3824.11,S,14435.81,E,015.60,235.00,030825,8.5,E,A*2C
```
And this is what is seen in Data Monitor.

[image]

Hakansv, 2025-08-03:
Does anyone know what this "Tag" is meant for?
I've read a document about NMEA0183, IEC 61162-1 v5.0, and can't find anything there. 
Since Dave,  @bdbcat, seemed to know about this, it's still something we should take into account.

transmitterdan, 2025-08-03:
> Does anyone know what this "Tag" is meant for? I've read a document about NMEA0183, IEC 61162-1 v5.0, and can't find anything there. Since Dave, [@bdbcat](https://github.com/bdbcat), seemed to know about this, it's still something we should take into account.

Normally it indicates the time the data was created, theoretically allowing downstream processing to put sentences from multiple sources into the chronological order or do other interesting things.

TwoCanPlugIn, 2025-08-03:
c Unix Time Stamps
r Relative Time Stamps
s: Source device information eg. Ss above by the OP,  the MX multiplexer description
t text 

Format for tags
\tag:tag data*hh\
checksum is the standard NMEA 183 xor checksum 

And it is possible to have multiple tags preceding the NMEA sentenes as well as tags without a sentence.

leamas, 2025-08-03:
> Don't we have any f.....g unit tests for any of the comms stuff ?

No. Patches welcome.

transmitterdan, 2025-08-03:
> [@Crosswind](https://github.com/Crosswind) wrote:
> 
> > Would it hurt removing this check entirely? Do you remember the exact purpose why this was introduced in the first place?
> 
> It wasn't me, so not sure. But I **think** this is about throwing away for example gibberish received when using wrong baudrate at an early stage, and as such I think it makes sense.

It's to satisfy the upstream expectation that a fully formed sentence will be sent by the lower level. The bytes from '$' or '!'  up to 'newline' is a fully formed sentence. At least it was when this code was first written.  It would not be hard to fix this on the serial I/O but then you have to do a lot of work upstream too.

TwoCanPlugIn, 2025-08-03:
I know Dave is vehemently opposed to a spec in whatever form.

Regardless of the absence of unit tests (personally I don't know how anyone can throw something over the fence without verifying it does what it says on the tin), can we have a brief confirmation of the following with respect to NMEA 0183. There is no point in reporting on the results of a unit test if you don't know what is the expected behaviour and what are the minimal release criteria.

And I'm kind of dumbfounded that the treatment of NMEA 1083 sentences differs when they are received by a serial port vs a TCP/IP port.  And even more astounded that we're having this discussion after how many releases ? 

What does OpenCPN do upon reception of:
Sentence does not begin with ! or $
Sentence does not terminate with CRLF <0x0D><0x0A>
Sentence includes a reserved character
Sentence does not have a checksum
Sentence has an invalid checksum
Sentence exceeds maximum length
Sentence has an unknown or unsupported sentence formatter
What are the supported sentence formatters
Sentence fails parsing depending on the sentence formatter 
Are encapsulated sentences (those beginning with '!' correctly formatted (eg. sentence number, total number of sentences, sequential message id, fill bits etc.)
Does OpenCPN support (ie. respond) query sentences (eg. Hakan uses these in his AIS Voyage Data plugin to query the AIS transponder)
Sentence is preceded by a tag block
if tag blocks are supported:
tag block lacks closing '\' character
tag block exceeds maximum length for the tag block
tag block has incorrect checksum (IIRC tag blocks must always have a checksum)
tag block includes an incorrect parameter code
tag block includes reserved characters

For output sentences
Are checksums enabled or disabled.
If a received sentence lacks a checksum is one created and appended to an outgoing sentence or is it transmitted without a checksum Is this affected by the enable/disable checksum option. 
Are received tag blocks stripped or are they included with outgoing sentences
Does OpenCPN append the FAA Mode to the generated sentences (most of the nav related sentences APB, RMB, RMC etc)

Just my 5' brain dump....

Hakansv, 2025-08-04:
Really an interesting and clear-sighted compilation of your thoughts. Admittedly, perhaps mostly for discussion, but relevant to this particular issue since the purpose should be to provide a basis for how OCPN should handle the "Tag" issue is about.

OCPN should handle the standard so that the program can be used for both new and old equipment mounted in different boats. "Tag" apparently occurs in some equipment and the OCPN should understand the sentences where it is found. Since we don't yet use this "Tag" for anything, the solution is to sort it out and use the rest, i.e. what starts with "!" or "$". Just like the proposed code change. That is the end of the question.

The FAA issue is interesting and admittedly does not belong to this issue, but briefly and without an invitation to further action right here. OCPN understands sentences both with and without FFA. At least for "RMC". However, the RMC OCPN produces, for APs, is without FFA because not all APs use FFA. Then we assume that the new AP also supports the old standard without FFA. It can change over time. So OCPN should perhaps broadcast two RMCs, with and without FFA? But as I said; That is not part of this issue.