OpenCPN/OpenCPN #4712: Route & Track Manager - failure to update navobj.db

Rasbats, 2025-08-16:
**Describe the bug**
Using Route & Track Manager to update the speed for a waypoint on the route fails to update all the waypoints. 
When the planned_speed is requested for a waypoint (PlugIn_Waypoint_ExV2) any waypoint that has not been updated shows a planned speed of zero.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Route & Track Manager
2. Find the route and click 'Properties'.
3. Scroll down to a waypoint and edit the planned speed
4. Query the waypoint with API 20 and the edited waypoint shows the new planned speed. Non-edited waypoints show '0.000000'

**Expected behavior**
The un-edited waypoints on the route should have retained the existing planned speeds.

**Screenshots**

[image]

[image]

[image]

[image]

[image]

[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Version 10

**Additional context**
Using wxPlugin_WaypointExV2ListNode to obtain the waypoint ExV2 properties.

TwoCanPlugIn, 2025-08-16:
I can't comment on your observations regarding planned speed, and the plugin API's, however

> And why does 'test2' have so many waypoints. Why was the database table/route not cleared before being re-written?

if you execute 

`SELECT COUNT() FROM routepoints;`

Do you get the number 48 (as indicated in the rows returned from your query above) ?

There's a bug in your SQL Query, well not a bug per se, as your query executes without errors, the problem is that it is the wrong query.

For example if you want to get the waypoint name and planned speed for the waypoints that constitute a route you'll need a more complex query, performing joins between the routepoints, routepoint_links and routes tables.

`SELECT point_order, routepoints.name, routepoints.plan_speed FROM routepoints LEFT JOIN routepoints_link ON routepoints_link.point_guid = routepoints.guid RIGHT JOIN routes ON routepoints_link.route_guid = routes.guid WHERE routes.name='test2' ORDER BY point_order;`

Rasbats, 2025-08-16:
DB Browser for SQLite...
Execution finished with errors.
Result: RIGHT and FULL OUTER JOINs are not currently supported

TwoCanPlugIn, 2025-08-16:
Works fine with the sqlite3 command line tool.

I wonder if DB Browser is compiled against the current SQLite version?

Rasbats, 2025-08-16:
Another bad from me. Updated DB Browser and voila. 
[image]
This shows the '0.000000' for unedited waypoint planned speeds.

Thanks Steve.

TwoCanPlugIn, 2025-08-16:
I'm going to start calling all of these classes of bugs, Fred Flintstone bugs. Yabba Dabba Doo; Yet Another DataBase Bug.

What's scary for an old curmudgeon like me, is that I'm digging out code snippets from at least 30 years ago.

I'll try and step through the code and find where the bug lies.

Were these waypoints upgraded (from navobj.xml), imported from GPX or newly created?

bdbcat, 2025-08-17:
Well, this is something of a conceptual bug, I suppose.
Has nothing to do with Flintstones database.

The planned speed of an individual routepoint in a route is generally undefined.  Recorded in database as "0".  In this case, the route calculation notices the "0" value for a routepoint's speed, and substitutes the parent route's planned speed in all ETA calculations and table preparations
```
    speed = pnode->GetData()->GetPlannedSpeed();
    if (speed < .1) {
      speed = m_pRoute->m_PlannedSpeed;
    }

```

Changing the planned speed of a particular routepoint does not affect other routepoints.  Speed for unedited points is still undefined, (0), and parent route's planned speed is used.

So, if one queries the planned speed of a particular routepoint, and finds planned speed is zero, then one should assume that the routepoint's planned speed has not been set, and so retains the default behavior.

There may be more intuitive means of expressing the routepoint undefined plan speed status.  Open to suggestions.
In any case, this has been the behavior of OCPN for many years, good or bad.
Not a regression bug.

Rasbats, 2025-08-17:
>finds planned speed is zero, then one should assume that the routepoint's planned speed has not been set

Can work with that. Closing.

Rasbats, 2025-08-17:
It would be nice if the variable m_planned_speed was available in the API. In the past it could be extracted from _navobj.xml._

```
speed = pnode->GetData()->GetPlannedSpeed();
    if (speed < .1) {
      speed = m_pRoute->m_PlannedSpeed;
    }
```