OpenCPN/OpenCPN #4713: OCPN v5.13: Context menus for objects no longer working

antipole2, 2025-08-17:
**Describe the bug**
When you hover over a waypoint, route or track, a yellow pop-up indicates you are over the object.
If you then control click, you should see the context menu for that object.
In 5.13.0-universal+c0be2e225 you only see the canvas context menu, as if you are not over an object.

**To Reproduce**
Try it

**Expected behaviour**
You should see the object's context menu or the canvas menu with the object's menu as a sub-menu.

 - OS: MacOS 15.6

**Additional context**
This is not concerned with the extra functionality introduced in 5.12
The basic functionality going wayback has disappeared.

My guess is that the processing of cntl-click now fails to properly check whether you are over an object

Hakansv, 2025-08-17:
For info: On Win and Linux a mouse right click does open the menus as before.

antipole2, 2025-08-19:
I confirm this issue fixed in build 5.13.0-universal+1e5229800

Thanks