OpenCPN/OpenCPN #4714: DIY satellite compass, support NMEA sentence THS!

bc38, 2025-08-19:
Now gaining popularity affordable GNSS RTK receiver high accuracy ( https://aliexpress.ru/item/1005008346420485.html?spm=a2g2w.orderdetail.0.0.62264aa6TU4l0d&sku_id=12000044679367566&_ga=2.57401071.497776179.1755613476-1117524455.1747227806 ), which gives the exact true course (0.05 degrees) with 20Hz, when using two antennas, in fact it is a satellite compass but with a price of $ 200. Also in this model ROT, Pitch, Roll is not calculated. These devices based on the modern UNICORE UM982 chip do not output NMEA messages in HDT, HDG..., but output True Heading data in the NMEA THS sentence. I would like to have the ability to support the NMEA THS sentence for the course (direction). I tested this solution, it works great with a distance of 35 cm between the antennas. The antennas definitely need an open sky, I tried to place them inside the car, the true heading did not work correctly.

Searching for information it turned out that THS is supported in FURUNO devices and ShipModul Multiplexers.

Now I have implemented support for THS => HDT via nmea converter: $XXHDT, $GNTHS1, $GNVTG2
and Pitch via the proprietary UNICORE sentence, HPR.
$XXXDR, A, $GNHPR3, D, PTCH, A, $GNHPR4, D, ROLL

bcn2, 2025-08-21:
THS:

from IEC61162-1
8.3.94 THS – True heading and status
NOTE This sentence replaces the deprecated sentence HDT.

Actual vessel heading in degrees true produced by any device or system producing true
heading. This sentence includes a “mode indicator” field providing critical safety related
information about the heading data, and replaces the deprecated HDT sentence.

$--THS,x.x,a*hh<CR><LF>
Mode indicator 1)
x.x   Heading, degrees true
a      Mode indicator 1)
Comments:
1) Mode indicator. This field should not be null.
A = Autonomous
E = Estimated (dead reckoning)
M = Manual input
S = Simulator mode
V = Data not valid (including standby)

Hakansv, 2025-08-28:
@All..
Discussion:
I'm on this. 
As can be seen from bcn2's post, there are several modes. 
For now my intention is to exclude all messages that do not have Mode indicator: "A". 
Thoughts?

bcn2, 2025-08-28:
Do see that as a correct interpretation.
If the Shipdriver plug-in would generate the sentence, then "S" would be correct as well.

bcn2, 2025-08-28:
Keep it with "A" only for sentences valid for OpenCPN. Used for navigation.

bc38, 2025-08-30:
works great
Thanks everyone!

[image]

bc38, 2025-08-30:
> Although no THS is seen in your Data Manager shot I suppose it's received to support the actual heading. Thanks for the report. Ready to close this issue?

[image]

closed 2025-08-30 by bc38