OpenCPN/OpenCPN #4733: grib_pi - Android sandbox makes loading grib files painful

DamianZaremba, 2025-08-31:
**Describe the bug**
Currently when trying to "Choose local GRIB file" only files within the app data directory/sandbox and an "emulated" downloads directory are accessible.

When a grib is present on the device though external retrieval it is difficult to access.

Currently a workaround is to perform "migrate chart data" on thr directory and then access the grib from the "migrated charts path", unfortunately this triggers a chart rebuild which is quite resource intensive.

**To Reproduce**
Steps to reproduce the behavior:
1. Download grib e.g. saved from email via a saildocs request
2. Open the grib plugin
3. Select the cog
4. Click on "Choose local grib"
5. Be unable to access the device paths to open the grib from step 1

**Expected behavior**
Being able to grant the app/plugin permissions to read the general file system would provide the easiest user experience.

Alternatively a "migrate" option similar to chart data would be workable (GRIBs are generally a lot more temporary then chart data so an easy way to remove them from the app data directory (not accessible via e.g. the "my files") would be needed if going down this path.

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: Samsung A25 5G
 - Andorid version: 15

**Additional context**
Files on device:
[image]

Max parent path accessible:
[image]

Workaround:
[image]

bdbcat, 2025-09-01:
For info, how do you do this:
"When a grib is present on the device though **_external retrieval_** it is difficult to access."

bdbcat, 2025-09-01:
OK, thanks.  Reasonably, it does not matter to OCPN where it came from, or how.  Just where it ends up.
I'm on it...

bdbcat, 2025-09-02:
Hmmm...
A little bit suspicious that the newest file "gfs....grb" is only 776 bytes.  This looks like the header only.  Where is the data?
But anyway, why is even the header file not visible to OCPN file chooser?

DamianZaremba, 2025-09-02:
> Hmmm... A little bit suspicious that the newest file "gfs....grb" is only 776 bytes. This looks like the header only. Where is the data? But anyway, why is even the header file not visible to OCPN file chooser?

It's a full file, just a very small (1 degree x 1 degree, gridded at 2 degree x 2 degree, for 3 time periods);
```
m00m00:Downloads damian$ wgrib gfs20250902102601318.grb 
1:0:d=25090206:PRMSL:kpds5=2:kpds6=102:kpds7=0:TR=10:P1=0:P2=24:TimeU=1:MSL:24hr fcst:NAve=0
2:86:d=25090206:PRMSL:kpds5=2:kpds6=102:kpds7=0:TR=10:P1=0:P2=48:TimeU=1:MSL:48hr fcst:NAve=0
3:172:d=25090206:PRMSL:kpds5=2:kpds6=102:kpds7=0:TR=10:P1=0:P2=72:TimeU=1:MSL:72hr fcst:NAve=0
4:258:d=25090206:UGRD:kpds5=33:kpds6=105:kpds7=10:TR=10:P1=0:P2=24:TimeU=1:10 m above gnd:24hr fcst:NAve=0
5:346:d=25090206:UGRD:kpds5=33:kpds6=105:kpds7=10:TR=10:P1=0:P2=48:TimeU=1:10 m above gnd:48hr fcst:NAve=0
6:432:d=25090206:UGRD:kpds5=33:kpds6=105:kpds7=10:TR=10:P1=0:P2=72:TimeU=1:10 m above gnd:72hr fcst:NAve=0
7:518:d=25090206:VGRD:kpds5=34:kpds6=105:kpds7=10:TR=10:P1=0:P2=24:TimeU=1:10 m above gnd:24hr fcst:NAve=0
8:604:d=25090206:VGRD:kpds5=34:kpds6=105:kpds7=10:TR=10:P1=0:P2=48:TimeU=1:10 m above gnd:48hr fcst:NAve=0
9:690:d=25090206:VGRD:kpds5=34:kpds6=105:kpds7=10:TR=10:P1=0:P2=72:TimeU=1:10 m above gnd:72hr fcst:NAve=0
```

DamianZaremba, 2025-09-02:
> So, thinking of your workflow. We may be able to add a public interface to OCPN that allows starting with an Intent() to effectively "import" a GRIB file. This would probably use the MediaStore API of Android. Your decoder app could call this Intent when needed. Lots of details to consider. Your thoughts on this?

I think a handler would probably solve this generically:
* For my workflow, I can use `MediaStore` which will give the option for OpenCPN with a correct intent
* For files from any generic app the "Share with" option should offer the same

Alternatively can the "Use local grib" option hop through a similar API to the chart migration logic i.e. copies the file into the sandbox and then load it?

I guess the question would be if this would act as an import in the sense of just rendering the file, or also persisting it into `Downloads/gribs` the same as "Download grib" does.

It's quite nice to be able to switch between files (e.g. different models), but on the flip side having to use a 3rd party app or add logic into OCPN for cleaning up old files doesn't feel super clear (perhaps this is the state of affairs today regarding the download GRIB functionality).

https://github.com/OpenCPN/OpenCPN/issues/4611 also proposed an intent for similar but different purposes (I'll look at that if I ever get to the end of making the compiling work on arm64 - 90% there but very time consuming)

bdbcat, 2025-09-02:
I would fashion the Intent as a true "import".  The file would be simply copied into the sandbox (GRIBs) directory already exists.  Might be useful to optionally open the GRIB_pi and display the file immediately.

Not sure what you mean by this:
" use a 3rd party app or add logic into OCPN for cleaning up old files "

https://github.com/OpenCPN/OpenCPN/issues/4611 is definitely related, back of mind enhancement I've been considering.