OpenCPN/OpenCPN #4737: Tide panel incorrectly synchronized with GRIB time selection

sebastien-rosset, 2025-09-03:
**Describe the bug**
The tide panel and GRIB plugin are currently linked together in a way that creates confusion during navigation. When a user selects a future time in the GRIB plugin to view weather forecasts, the tide panel also changes to show tide information for that same future time, rather than maintaining independence to show current tide conditions when needed.

Filed on behalf of @plang732. Pierre, feel free to update the description of this issue as you see fit.

**To Reproduce**
Steps to reproduce the behavior:
1. Open OpenCPN with both GRIB and tide functionality enabled
2. Open the GRIB dialog and move the time slider to a future date (e.g., 5 days ahead)
3. Open a tide dialog for your current position
4. Observe that the tide dialog shows tide information for the future date selected in GRIB, not current time
5. To see current tide conditions, user must manually navigate back in time using Prev buttons
6. This causes loss of the future weather forecast view in GRIB

**Expected behavior**
The tide panel should operate independently from the GRIB time selection. When a user opens the tide dialog, it should default to showing current tide conditions regardless of what time period is selected in the GRIB plugin. Users should be able to view current tides while maintaining their selected future time in GRIB for weather planning.

**Screenshots**
User mentions seeing a "blue line" in the tide panel instead of the expected "red line" indicating current time, leading to navigation errors.

**Desktop (please complete the following information if applicable):**
- OS: Windows
- Version: OpenCPN 5.12.x and later versions

**Additional context**
This issue is particularly problematic for sailors who need to:
- Check future weather conditions for passage planning (GRIB)
- Simultaneously monitor current tide conditions for immediate anchoring decisions
- Work from a distance (user mentions being 4 meters from display at steering wheel)

This issue is independent of the timeline widget proposal in PR #4578

Arne2024, 2025-09-04:
Please consider before making a change: 

I believe that the current version with the feature that links GRIB with panels for current and for tides as a very powerful tool, and should be kept.

Blue lines and blue current icon or arrow show times taken from GRIB and red line and orange current arrow means not linked to GRIB. This is easy to learn and not confusing in my view.

The synchronisation of both the tide and the current panels with a GRIB is a very elegant and powerful feature of OpenCPN.  It allows to run scenario's and prepare routings and compare GRIB data (is it any good?) with already available tidal and current from the harmonics that are stored for the are.

I would not be in favour to stop providing this feature, but maybe it can be added what the meaning is of blue and red lines, and of the colour code in the current arrows.  The blue line shows the simulated - deviation from the red line from machine time.  

Maybe the solution can be to only make the coupling with tide and current panels when Current GRIBS are in use, and not for weather GRIB?

Arne2024, 2025-09-04:
maybe it's a suggestions to explain the link with GRIB and the current and tide panels, and the colour coding used, in the manual. As it is a useful feature to easily check upcoming tide and current events by loading a grib files when preparing a routing or a navigational plan. And offer some screenshots and examples in the manual.

sebastien-rosset, 2025-09-04:
> Please consider before making a change:
>
> I believe that the current version with the feature that links GRIB with panels for current and for tides as a very powerful tool, and should be kept.
>
> Blue [...]

All good points. In retrospect, I should have let @plang732 create the issue, I only created this issue on his behalf. Pierre has very strong opinions about this. Hopefully he will comment and propose a well articulated solution that works for everyone.

Wrt to the two vertical lines, thanks for providing the feedback (I made some fixes and enhancements in #4664). I do think further usability enhancements can be made. For new users, it's not intuitive that the time controls are from the GRIB plugin as opposed to a more central/core widget.

rgleason, 2025-09-04:
@sebastien-rosset  Maybe the title should be changed as it is a bit misleading. "incorrectly"??  I happen to like how it works, but I need to pull and compile again and play with it.

Can there be a hover function over the red and blue, identifying what it is? or does that conflict with other possible operations?

plang732, 2025-09-04:
I can understand that some sailors like the link between the Grib and Tide panels (and maybe another feature as Currents, etc.) 

This is the demonstration of what I wrote before : there are many different times : Grib, Tides, Currents... and the temporary or variable needs of the skipper (routing study, tricky passage sailing, anchoring, etc.). Sometime it is useful to get them linked together. Sometime it is not.

So I would like OCPN to remain or be opened to all those needs by implementing easy to develop and understand options.

I suggest to gently improve the Tide dialog the following way. It will satisfy both @plang732, @Arne2024 at 100% and probably many other users the same way, without adding any complexity to OCPN.

Remember, there is a [Now] button in the Grib dialog, but not in the Tide panel.

** Two new controls in the Tide panel**
- Add a [Today] button between the [Prev] and [Next] buttons.
- Add a <Follow Grib time> checkbox above or under the 3 buttons. Arne2024 will check <Follow Grib time> and Plang732  will uncheck it. Everybody will be happy :-)

**When the [Today] button is pressed**  (by Arne2024 or Plang732) 
- The Tide panel will show to the current date and the red line for the current time.
- The <Follow Grib time> will be unchecked by the programmer to avoid time conflicts between Grib and Tide. So the two panels will be disconnected and independent from each other (as Plang732 likes it !)
- I would not advise to link the [Now] button of the Grib and the [Today] button of the Tide panel. Its is easy to press a button.
- Note : This [Today] button will be very useful even if the Grib dialog is closed to avoid pressing [Prev] many times to come back to the current time.

**When <Follow Grib time> is checked**
- The tide panel will show immediately the tide for the time selected in the Grib dialog.
- If the time in the Grib dialog changes, the Tide panel will always follow (as it is in the case in current OCPN version 5.12.2)
- The OCPN settings (opencpn.ini) can eventually remember the last state of this checkbox, for the next time the dialog is open.

**The blue line** behavior should remain the same as today. It was a brilliant idea.

Remember : Weather forecast, Tides, Currents, Skipper needs, etc. are independent concepts. Some of them are temporarily linked together by the skipper **depending on his current activity** (which is highly variable following what he is doing at the chart table or at home).

I think this two new dialogs controls will be very easy to implement and it does not need any documentation to allow the users to understand them.

Regards, Pierre

sebastien-rosset, 2025-09-04:
Pierre, feel free to edit the title (hopefully you can). This is your request, not mine. I just wanted to decouple this particular ask of yours from the timeline widget, as they are two distinct work items.

sebastien-rosset, 2025-09-05:
> Slightly related (I can open an issue but figured I'd tag on here) ... I often have multiple tide windowd open when planning a trip. I see that has been disabled. Maybe an option (shift-click) to allow for multiple ones again?

This was done to fix the issue identified here: https://github.com/OpenCPN/OpenCPN/issues/4663#issuecomment-3118000622. Having a single window was a simple way to fix the issue because there is a singleton field that was causing a memory leak. With some refactor, it should be possible to properly implement multi window or tabbed panes.

Arne2024, 2025-09-05:
Thank you for considering to keep the link with a time driver (now the Grib, which works in most cases, and can be linked or unlinked.

I agree with comments that, ideally, there could be a system driver in the that can drive time of current (independent of an available grib with useful time)   

For now a button seems a good solution: maybe name it ''Link to grib time / Link system time? 

About multiple tide panels: I also feel that the option to have multiple Tide Panels (and Current panels) open at same time valuable.

And when planning like to see Current and Tide Panel at same time, which is now possible when driven by GRIB time,
But again,  ideally they could be linked to a panel like the OTcurrent driver panel, but OTcurrent is not linked to the panels for tide and current.

I attach screen shot with OTcurrent and tides around Brittany as example. 
 ￼

Thanks for good discussion.

Arne

I can't comment on the most robust solution, but an option could be to have the same panel that drives the o-current plugin also drive the Tidal Panel. now the link with crib time robust  , I believe they work exactly the same)

> On 5 Sep 2025, at 08:29, Sebastien Rosset ***@***.***> wrote:
> 
> This was done to fix the issue identified here: #4663 (comment) <https://github.com/OpenCPN/OpenCPN/issues/4663#issuecomment-3118000622>. Having a single window was a simple way to fix the issue because there is a singleton field that was causing a memory leak. With some refactor, it should be possible to properly implement multi window or tabbed panes.
> 
>

Arne2024, 2025-09-05:
> plang732
>  left a comment
> (OpenCPN/OpenCPN#4737)
> Thank you, Pierre for the thoughtful suggestions. I think the addition of the Now and Today button and the functionality checkbox is a reasonable idea. What would you call that checkbox? Probably not the Pierre-Arne checkbox, but we need a name that explains it. Maybe a check in the checkbox should "Disconnect Tide Panel"? or what?
>
> Disconnect Tide Panel => disconnect from what ? I think it is not self explanatory enough.
>
> I would prefer or . Positive or negative as you prefer. With or without the word "time" as we know it is not disconnect from some fruit juice ! So the user will know to what is the possible connection.
>
> I suggest also to evaluate if the Grib dialog could be something like a kind of master time module to which other modules like Tide can connect or not. As far as I know, Tide was the first or is the single module to be linked to the Grib dialog.
> Maybe current, but he has no panel, I think (only the Show current in the popup menu) ? I have no current data in the areas where I am sailing since years (Europe between Greenland and Greece).
>
> Or, maybe the TimeLine will implement this master time so that any other time related panel could connected itself or not. If each of these modules can choose independently if it connects to the TimeLine or not (with a simple checkbox), it could probably be nice. Maybe with an option somewhere to connect or disconnect all time related modules. Sebastien will evaluate this better that I can. This is my limit of knowledge.

H

I am sure you are aware, just in case not, there’s a resource with useful current harmonics , merak V10 which covers tides and current in some key areas from North Sea coast down to the Channel and Normandy and Brittany but also Netherlands and Germany,

For France Atlantic and Manchester,  there is an excellent pi , frurrent, with  detailed tidal streams and a time driver which includes all  major tidal areas, with frcurrent, you can download the harmonics with the plugin, they drive from the data set that comes in Merak v10 file.

See a screenhot of French tides around Ils ‘ousseant and Chenal du Four, the orange arrows and tidal panel do not link with the frcurrent driver, so just for reference, and the red line is the local time of the pc used in this case..

Regards,
Arne

Arne2024, 2025-09-05:
> I am only aware about it very superficially. I live in the United States. Is there something about it that should be considered when designing for this enhancement or for #4578 <https://github.com/OpenCPN/OpenCPN/pull/4578> ?

No, I think these harmonics for tide and current are doing the same as your US harmonics, and just doing the same and what is expected.

This remark was for who mentioned sailing between Greece and Greenland and not having currents (pierre or David?).

Arne2024, 2025-09-05:
French currents from FRCurrent pi and tide and current panels shown

[image]

occurrent plug-in showing tide and current panel , in this case at local PC time, not linked to the otcurrent pi.

[image]

plang732, 2025-09-05:
@sebastien-rosset,

> You created this thread "for me" as you considered that the tide panel should not be discussed into the TimeLine one.

So, it is not a good idea to import the #4578 here !

Why the existing structure of OCPN (which is not respected by the Grib module if I understood well) should influence the user interface ?

You are focused on the source code, not on the user needs. A good application should focus on the user needs and easy to use, not on the programming architecture, the programmer feeling or give a meaning to the life after being retired from the business.

You want to improve and renew OCPN to version 5.14. We do not know when such version will even be released (with or without your timeline)

Here we discuss about version 5.12.3 which should be a minor version of v5.12.0 and v5.12.2.
Remember that you choose the title "Improve time control actions in tide panel" to avoid speaking about a bug.

I am not surprised that you expect to "improve" it by adding a TimeLine or another type of central time system. 

> If you consider new users for a minute, 

It is a nonsense to speak about a possible new user. Why a possible "new user" should trigger the evolution of a software which is already used by 100000 users ?

> It's not intuitive that these time control buttons are part of the GRIB panel. 

Are you really serious ???????????????????????

At the moment, the Tide Panel IS PART of the Grib one. Is it intuitive ?
How a user can feel that to come back today in the Tide Panel, he has to press [Now] in the Grib window ?

What I ask is to unlink it from the Grib panel !!!!!!!!! 
Your dream is to look at the chart. But when I open the tide panel TODAY, it is to get more details than the small icon displayed on the chart. 

**My previous suggestion will be 100% intuitive, for all users present and future, for some minutes programming !**

It is not the place, here, to plead for you TimeLine project !!!!
Nor to discuss about the simulation of the weather/tide situation while studying a route... (I think there is maybe a routing plugin for that)

It looks more and more that your design strategy is **why making it simple when it can be made complicated** This give of course a meaning to the programmer work. Please return to the #4578 thread or another one that you create for yourself to discuss the old and new architectures of OCPN !

If you are the new OCPN team leader and you replaced or try to replace Dave, say it to the world. It will be helpful to discuss with you ;-)

I gave up in the thread #4758. 
I am sorry. I regret you reaction after you asked me to suggest a solution... 

Thanks

Arne2024, 2025-09-05:
I think the application works well very well and the grib time driver is a powerful feature.
A button to link or de-linnk to the grib would be an enhancement.  

Attached 3 shots made with the most recent Irish Grib and our own tides and currents for stream station in the Channel (Chanel 1) and for tide at port of Cherbourg.

The Grib and the Tide and Current panels are working as what I think most users would except. 
Red is Now, Blue is time in the GRIB.   
If you want revert to Now only just close the grib.

[image]
[image]
[image]

Arne2024, 2025-09-05:
Hi, I will be happy do, I will try to clean it up and send you shortly. I assume ion text box like this one?

sebastien-rosset, 2025-09-05:
@rgleason , I think I see the problem.

The master branch header file does declare `std::list<wxPoint *> m_sList`, i.e. some changes have been made to migrate to STL containers.
However, the change seems incomplete: the `OnTCWinPopupTimerEvent` method still uses the old casting approach `(wxPointList *)&m_sList`. This is a bug in the master branch - it won't work correctly because you can't safely cast a `std::list` to a `wxPointList`. I'm testing a fix locally.

sebastien-rosset, 2025-09-05:
> Thanks Arne, use the link and add your text and links in a comment box. --Very helpful.
> 
> [#4743](https://github.com/OpenCPN/OpenCPN/discussions/4743)
> 
> Just downloaded version 5.12.2 and it does not have the red and blue bars, which was the result of an intentional PR to remove the feature. We are hoping it will be revised and added back.

I don't think there was any intent to remove the red and blue bars, at least not that I am aware of. It was caused by an accidental regression with the removal of the `WX_DEFINE_LIST` code. The good news is that there is a fix.

rgleason, 2025-09-05:
Note: I think my statement

> Just downloaded version 5.12.2 and it does not have the red and blue bars, which was the result of an intentional PR to remove the feature. We are hoping it will be revised and added back.

It does show red and blue lines when I activate Grib_pi.

[image]

Arne2024, 2025-09-06:
Hi, just to make sure: the above example show a current and a tide panel opened at same time and moving along tide and current, not 2 tide panels:
1.  - the top panel Calais = Tide 
2.  - bottom panel  Pas de Calais = current. 

I find side-by-side current and tide very useful for planning.
So hopefully we to keep the option to show at any same time a **tide** panel AND a **current** panel besides the grib reader. 

I need to think about your question
> Do you have an opinion on whether it should be two separate tide panels like it has been up to version 5.12? I.e, you can move the panels independently?

I think it may be useful to have a 2nd tide (or 2nd current) panel, but not sure if they should also be linked, or kept independent, maybe the latter?

hope this helps.

rgleason, 2025-09-06:
Pierre,

> I would be happy with independent panels, but not with the link with the Grib panel...

We are trying to keep the operation simple with as few controls as possible so users don't have to consider many options.
There is no promise in all of this. At this point we are trying to understand what you think you really need.
Engineering something that works for all may be possible, but it has to be logically simple.
So do I understand you correctly, that you want (one or multiple?) Current and Tide Panels that are independent from Grib_pi?

1. One Current panel and one Tide panel that is independent from grib_pi?
2. Two Current panels and two Tide panels that are independent from  grib_pi? So you can compare two different locations.
3. Multiple Current panels and Multiple Tide panels that are independent from grib_pi?
4. Can the panels be forced to be only from different locations for Tide Panels?
5. Can the panels be forced to be only from different locations for Current Panels?
6. Do you ever have a need for 2 tide or current panels from the same location?

What do you mean by independent from Grib_pi? What do you mean by a link with Grib_pi?
Please explain as clearly as possible so that Sebastien and I understand.
Your needs are not satisfied by having Tide and Current panels that do not show the blue line when grib_pi is not open?
I thought you said you seldom use grib _pi? 

I did not forget the Today Button or the Disconnect from Grib button (which would be in the Tide or Current Panel?).
What is the disconnect from Chart button for, I haven't thought about this as much as you have?

Please keep in mind that Sebastien is going to do the best he can, but this has got to be useful and simple without too many choices which confuse everyone.

Please detail exactly what you need and would like to see. There will probably have to be some compromises however.

rgleason, 2025-09-06:
Keep in mind what Sebastien explained and wrote:

> IMHO, a better approach is to extract time management into a core time service that plugins can subscribe to. This would allow each time-dependent module to independently choose its connection to the time service through a simple checkbox interface. I want to make it clear these programming considerations do not necessarily change the UI functionality.

rgleason, 2025-09-06:
Ensure single instance of tide dialog #4668     https://github.com/OpenCPN/OpenCPN/pull/4668

Multiple instances cause memory leaks, which we can't have. So multiple panels appear not to be possible. Sorry.

sebastien-rosset, 2025-09-06:
> Ensure single instance of tide dialog [#4668](https://github.com/OpenCPN/OpenCPN/pull/4668) [#4668](https://github.com/OpenCPN/OpenCPN/pull/4668)
> 
> Multiple instances cause memory leaks, which we can't have. So multiple panels appear not to be possible. Sorry.

Multiple tide panels are possible, but that would require significant refactor to support  eventing going from central logic to every tide panel and still avoid memory leaks.

With that said, I was looking at PR #4426 and wondering what would be best behavior that would work for everyone. That PR is about adding a date/time picker in the tide panel. 

1. Single tide panel that can show multiple stations (most likely stacked vertically). In that case the question is whether there would be a single date/time picker for all stations.
2. Multiple tide panels, each with their date/time picker. In that case if you actually intend to see the tide level of the stations at the same time , then potentially it could be a bit tedious because you would have to pick the same date/time in all panels. The flip side if if you want different date/time then you have a bit more flexibility.

rgleason, 2025-09-06:
**Regarding #1 above**  
1. Single Tide Panel for multiple stations. - Might be worthwhile. -Would Currents be different or the similar?
2. Date/Time Picker - single one for all stations, sounds good, not sure how it would work.
3. So this version is best for showing multiple "different" tide stations.
4. Would this avoid the memory leak?

**Regarding #2 above**
1. Multiple tide panels each with a date/time picker. Presumably you can pick different locations and set the same time.
2. This one is easier if you want different times with the same station or two different ones.
3. Would this avoid the memory leak?

**Also**
1. Would there be a dropdown selection for  Timeline, Grib, ETD, ETA or perhaps month/day/year?
2. Would there still be room for Prev, Next, and Today? - I know that some people just wanted the buttons bigger for their finger.
3. Regarding disconnect from Grib or Chart, perhaps instead, it is just a checkbox "Show blue Line" or "Don't show blue line" to satisfy Pierre's need.

I am still trying to understand the whole picture here and what is needed, so I don't have a strong opinion yet.

davidsanner, 2025-09-06:
Lots of options for multiple tide stations displayed, not even sure what I would say is the best option but having some way to display multiple ones would be nice for my typical use case.

For example when planning a sail where the tides change significantly at each station I typically would open 3 or 4 windows, set each of them to the same day and then evaluate estimating the time I might arrive at each station.  I do this even when I'm planing a sail (or race) days out. 

If the solution was a single tabbed window it might be nice to have them all show the same date and then be able to quickly use the keyboard tab to switch between stations.   An option to unlink date on each tab might be a fancy option but not so important for how I use it / imagine typical use.

rgleason, 2025-09-06:
Actually I think it is useful to set different days too, but the main use case would be similar to what you describe. This reminds me that Rasbats has a plugin that tries to do just that for a given route, finding the nearest stations at each WP with corresponding time given a certain speed. I found it to be pretty good when I tested it. 

Earlier I was thinking that there really is not use for more than two such windows and I was going to suggest to Sebastien that we just start with two windows. Now I am not so sure, can you play with this in your mind and with the tide tool that we have now and try to determine what the bare minimum would be, because you would be able to progressively change the tide location to accomplish the same task without using quite so much screen space...