OpenCPN/OpenCPN #4741: Using Tide Panel with Grib pi open -crash

rgleason, 2025-09-05:
@sebastien-rosset 

**Describe the bug**
Pulled current master. Cleaned and Rebuilt. Started Debugging.
Opened Grib_pi, moved to east coast RI-CT-Fisher's Island , opened recent grib of that area. 
Turned on Tides and Currents.
Right clicked on  Race Rock (SE of Fisher's Island) 
Tide chart opened and both Red and Blue lines exist.
Then the Debug window flashes up. 
See screenshot please. 
The second screenshot is the Opencpn screen  all I can do is move around a little red X.
Then back to the debug window, click continue and OpenCpn closes. 

**Expected behavior**
Expected it to work, so I could use and see the new features in Tides.

**Screenshots**

[image]

[image]

**Desktop (please complete the following information if applicable):**
 - OS: current master  5.13.0
 - Version  5.13.0  Windows 11

**Smartphone/Tablet (please complete the foll
No

Do I need to update the API somehow? or perhaps opencpn-libs or something?

rgleason, 2025-09-05:
Then tried  cleaning the build and building again. Then did the same procedure.

[image]

[image]

[image]

[image]

[image]

rgleason, 2025-09-05:
Again just to see if I've done something wrong..
Find this exception is thrown

Exception thrown at 0x65BA8E77 (wxbase32u_vc14x.dll) in opencpn.exe: 0xC0000005: Access violation reading location 0x8C02B40C.

[image]

[image]

[image]

rgleason, 2025-09-05:
Is this a wxwidgets problem?

bdbcat, 2025-09-05:
No, most likely trouble in the master branch due to some fairly serious refactoring of how "lists of things" works in OCPN.
Probably results from adding the Tide/Current pick-points to the waypoint list.
Does it crash if you do not open the Current graphic dialog, but just pan around with the current arrows shown on chart?

rgleason, 2025-09-05:
Do I have to rebuild wxWidgets or something?

rgleason, 2025-09-05:
Using TransmitterDan's procedures, when I did these commands, OpenCPN work

-  cd OpenCPN
-   .\buildwin\configdev.bat --debug --relwithdebinfo
-  cd .\build\relwithdebinfo
-  .\opencpn -p

And when I open a tide or current panel it immediately fails.

So I think my MSvisualStudio is working correctly and correctly shows the failures.

rgleason, 2025-09-05:
Ok have now rebuilt and cleaned several times and now it is working better, and I get similar results and crashes to the examples above.

[image]

[image]

[image]

**COPILOT** 

The exception is an access violation (0xC0000005) when reading memory at `0x8805430C`, triggered in `wxbase32u_vc14x.dll` during a call to `wxPointList::Item(idx)`. This typically means a null pointer or invalid memory access.

From the code, the likely immediate cause is that `idx` is out of bounds for the `wxPointList` (`m_sList`). If `idx` is greater than or equal to the number of items in the list, `Item(idx)` will return `NULL`, and dereferencing it with `->GetData()` will cause an access violation.

Root cause analysis requires knowing:
- The value of `idx` at the time of the exception.
- The number of items in `m_sList`.
- Whether `m_sList` is properly initialized and populated before this access.

Without this data, the most probable root cause is that the logic for calculating `idx` does not guarantee it is within the valid range of the list, especially if `m_graph_rect.width` or `curs_x` have unexpected values.

To fix: Add bounds checking before accessing `Item(idx)` and `Item(idx-1)`. For example:
```cpp
if (idx > 0 && idx < m_sList.GetCount()) {
    wxPoint *a = list->Item(idx - 1)->GetData();
    wxPoint *b = list->Item(idx)->GetData();
    // ... rest of code
}
```
If you can provide the values of `idx` and `m_sList.GetCount()` at the time of the crash, I can confirm this diagnosis or look deeper.

sebastien-rosset, 2025-09-05:
@rgleason It would be awesome if you can test on Windows now that the fix has been merged. I've tested on Mac. If you have cycles to test on Windows, that would be great.

rgleason, 2025-09-05:
@sebastien-rosset  Thank you Dave and Sebastien. I am back and will test now. and will advise.

I next need to figure out how best to test your PR's so that I can try them out and learn.
Should I just fork your opencpn  to a new local directory like source/opencpn-sebastien directory and then point MSvisualStudio to that directory and build? Is it that simple since I have opencpn  "master" branch compiling.

Also I guess I would have to go to the right branch for the purpose and pull that down and build that?

sebastien-rosset, 2025-09-05:
> [@sebastien-rosset](https://github.com/sebastien-rosset) Thank you Dave and Sebastien. I am back and will test now. and will advise.
>
> I next need to figure out how best to test your PR's so that I [...]

You don't need to create separate directories or fork again. You can test different PRs in your existing local workspace by adding the contributors' **forks** as additional remotes. Here's how:

Instead of forking to separate directories, you can add my fork (and others) as remotes in your existing OpenCPN directory:

```bash
cd your-existing-opencpn-directory
git remote add sebastien https://github.com/sebastien-rosset/OpenCPN.git
git fetch sebastien
```

Now you can easily switch between different branches/PRs:
```bash
# Test my PR branch
git checkout sebastien/my-pr-branch-name

# Go back to master
git checkout master

# Test another contributor's work
git remote add contributor-name https://github.com/contributor-name/OpenCPN.git
git fetch contributor-name
git checkout contributor-name/their-branch
```

For example, in my workspace, I have the following forks:

```bash
$ cd OpenCPN
$ git remote -v
leamas	git@github.com:leamas/OpenCPN.git (fetch)
leamas	git@github.com:leamas/OpenCPN.git (push)
origin	git@github.com:sebastien-rosset/OpenCPN.git (fetch)
origin	git@github.com:sebastien-rosset/OpenCPN.git (push)
upstream	git@github.com:OpenCPN/OpenCPN.git (fetch)
upstream	git@github.com:OpenCPN/OpenCPN.git (push)
```

sebastien-rosset, 2025-09-05:
Great to hear! Maybe you could upload screenshots showing the formatting issues.

rgleason, 2025-09-06:
https://github.com/OpenCPN/OpenCPN/issues/4745

PR #4667 (2025-09-05): Bug fix: Tide dialog station text overlaps chart area when window is resized smaller

## Changes in this PR

- Fix #4666
  - Modernize TCWin layout with sizer-based layout.
  - Replace hardcoded positioning calculations with automatic layout management.
  - Eliminates text spillover and overlapping panels when resizing the window.
  - Create custom TideChartPanel class for self-contained chart rendering.
- Fix #4741

## Screenshot

When the user resizes the tide dialog, the dialog uses responsive UI to properly render and resize the sub-elements. Previously, some of the text areas were spillin [... cut]