OpenCPN/OpenCPN #4750: Compiling OCPN 5.12.x / Windows 11 - WinConfig.bat and VS are failing

plang732, 2025-09-07:
Hi,

After downloading the master source (as *.zip or with git) from https://github.com/OpenCPN/OpenCPN we are expected to launch the WinConfig.bat script. This script :
- Has bugs.
- Displays unreadable warnings and errors messages (does not pause after display)
- Does not give any feedback when failing (it closes the cmd window without pause)
- It creates strange directory as "installed"
- It is running "pushd cache" but the cache directory does not exist (line 466).
- Etc.

**Some example running WinConfig.bat**

> Installing pre-commit hooks
>Requirement already satisfied: pip in c:\dev\opencpn-5-12-x\buildwin\python.3.13.7\tools\lib\site-packages (25.2)
>An error has occurred: FatalError: git failed. Is it installed, and are you in a Git repository directory?
>Check the log at C:\Users\plang\.cache\pre-commit\pre-commit.log
>'.' is not recognized as an internal or external command,
>operable program or batch file.
>Error: Could not find Python and/or pre-commit tool.

> Display in RED color
>Creating library ..\..\lib\vc_dll\wxmsw32ud_qa.lib and object ..\..\lib\vc_dll\wxmsw32ud_qa.exp
>C:\Dev\ocpn_wxWidgets\include\wx\msw\glcanvas.h(18,10): error C1083: Cannot open include file: 'GL/gl.h': No such file or directory [C:\Dev\ocpn_wxWidgets\build\msw\wx_gl.vcxproj]
>C:\Dev\ocpn_wxWidgets\include\wx\msw\glcanvas.h(18,10): error C1083: Cannot open include file: 'GL/gl.h': No such file or directory [C:\Dev\ocpn_wxWidgets\build\msw\wx_gl.vcxproj]

Then it exists without notice...
The build directory is empty...

**VisualStudio**
VisualStudio cannot build OCPN, even if, by chance, opencpn.vcxproj and opencpn.sn are created
- Does not find "dnet.h"
- Cannot build "libdnet" as several "*.rule" are not created at build time.
- Etc.

Note : In June/July, I could compile it with some minor fix, but it was OK. And I gave access to Dave (AnyDesk) to fix bugs in the the beta 3 versions before v5.12.0 was released.

Since that time, it became to be impossible to reinstall & compile the sources

Regards, Pierre

leamas, 2025-09-07:
> There are dnet.h files in two locations

Which is as expected. The build first downloads pristine sources and in a later step creates a patched copy of them, 

> Nice to know that so many user are compiling Opencpn without problem ;-)

Yes,  isn't  it :). Even nicer is the fact that you are the only one reporting these problems, so we know for sure that what **might** possibly be bugs is only exposed in your  environment.  

The basic form for a bug to be accepted is that you show us a reproducible way to trigger this bug. So far you have not done this, the workflows you describe without much details  works for others. 

Hacking the build trying to try to handle the problems you have is not the right way, other user reports  that they don't need to so you should not either. The problem is to adjust you environment so it matches what for example @nohal  or @Hakansv  have so you get the same results.

plang732, 2025-09-07:
Let's see what happen to @Hakansv

plang732, 2025-09-07:
> The basic form for a bug to be accepted is that you show us a reproducible way to trigger this bug. So far you have not done this, the workflows you describe without much details works for others.

Lol ! I explained everything :-)
Remove every software with Revo Installer.
Install VS using the link from the Opencpn documentation (link given here above)
Install git the same way
Reboot the PC
Start the "developer command prompt"
Run winConfig.bat

Look at the results which you did not read from my previous post.
Etc...

Anyway, now I have a running compiled version...

Thanks a lot for your help...

rgleason, 2025-09-07:
Pierre, so what was your special magic? What made it work for you? 
Try to leave a few suggestions for others to follow this road, this will be much appreciated by those who follow!
Thanks!

PS: this is linked to a new General Discussion area topic  Windows MSVC Builds (Specific Process) #4751   https://github.com/OpenCPN/OpenCPN/discussions/4751

rgleason, 2025-09-07:
BTW Pierre  since there is no specific bug, and you have succeeded in compiling Windows OpenCPN ,  and you are the OP (Original Poster), you should close this!

plang732, 2025-09-07:
>BTW Pierre since there is no specific bug, and you have succeeded in compiling Windows OpenCPN , and you are the OP (Original Poster), you should close this! ..and congratulations. I hope to see you testing PR's and helping Sebastien with the finer details of Tides, Currents, Timeline, etc.

OK. I will close this thread, BUT I still claim that there are bugs and a wrong design in winConfig. For sure... There are folder which are not created, invalid BAT instruction, etc. My computer settings cannot be the cause ! How my computer settings could trigger errors in opencpn.sn ? Anyway, I will not fight (fire ;-) more...

It is not the single issue. OpenCPN is crashing while using the Intel/OpenGL driver. It cannot be a bug made by Intel as other OpenGL software do not crash on my PC. But it is said as in the current thread, that the problem is not in Opencpn... but in the Intel driver !

Anyway... I would like to try the timeline...

Today, I compiled the version 5.12.2 from the master. It looks that it is starting normally.

Now, the challenge will be to compile the project of Sebastien.

My expertise in VS, Git, and the way Opencpn is patched, updated and compiled is near 0.
I am just an old successful developer of commercial Windows software (1990/2001). I help Dave to fix bugs in the early times of OpenCPN and in last July. It's what I can do...

Can you send me some suggestions or tricks to compile the 5.13 version of Sebastien ?
Do I need to git HIS master (from where ?) and follow the same way I worked around winConfig.bat to compile it today (as described in this thread) ?

Regards, Pierre

rgleason, 2025-09-07:
Pierre, this is good. We can continue using this Issue, while it is closed.

**Recording what you did**
Yes there may be some issues with winconfig.bat because the environment changes over time and everything gets updated.
TransmitterDan has been maintaining this Compile feature, but he needs help learning what needs to be fixed.
So if you would, please describe what you found to be the one or two things that you did that has made it compile acceptably, if you can remember.

**Moving forward**
I has the same question which Sebastien answered here
See https://github.com/OpenCPN/OpenCPN/issues/4741#issuecomment-3260012562
Also please see https://github.com/OpenCPN/OpenCPN/pull/4578#issuecomment-3263214711 and further down.

We need to make a summary page of this for reference and others!
I haven't learned how this was done for 15 years! I kept hitting stumbling blocks on other stuff and never got to it.

rgleason, 2025-09-07:
Also you have to set up your remotes mine are like this

C:\Users\fcgle\source\opencpn>git remote -v
origin  git@github.com:rgleason/opencpn.git (fetch)
origin  git@github.com:rgleason/opencpn.git (push)
sebastien       git@github.com:sebastien-rosset/OpenCPN.git (fetch)
sebastien       git@github.com:sebastien-rosset/OpenCPN.git (push)
upstream        git@github.com/opencpn/opencpn.git (fetch)
upstream        git@github.com/opencpn/opencpn.git (push)

The remote commands that you can use are
git remote -v
git remote rename <old-name> <new-name>
git remote rm origin
git remote add origin https://github.com/username/reponame.git

However I use this because I have set up ssh - Look on github how to do that.
Change to ssh: git remote set-url origin [git@github.com:rgleason/](mailto:git@github.com:rgleason/)<plugin_pi>.git
Change to https: git remote set-url origin https://github.com/USERNAME/REPOSITORY.git

3. Configure remotes
When a repository is cloned, it has a default remote called origin that points to your fork on GitHub, not the original repository it was forked from. To keep track of the original repository, you need to add another remote named upstream:
A remote is a repository stored on another computer, in this case on GitHub's server. It is standard practice (and also the default when you clone a repository) to give the name origin to the remote that points to your main offsite repository (for example, your GitHub repository).
Git supports multiple remotes. This is commonly used when forking a repository.

    $ cd opencpn302-git
    # Changes the active directory in the prompt to the newly cloned "opencpn302-git" directory
    $ git remote add upstream https://github.com/rgleason/opencpn302.git
    # Assigns the original repository to a remote called "upstream"
    $ git fetch upstream
    # Pulls in changes not present in your local repository, without modifying your files

With >1 Remotes
Before you can start working locally on a remote branch, you need to fetch it as called out in answers below.
To fetch a branch, you simply need to:
git fetch origin
This will fetch all of the remote branches for you. You can see the branches available for checkout with:
git branch -v -a
With the remote branches in hand, you now need to check out the branch you are interested in, giving you a local working copy:
git checkout -b test origin/test

https://stackabuse.com/git-fetch-a-remote-branch/

    git fetch <remote-repo> <remote-branch>:<local-branch>
    $ git checkout <local-branch>

rgleason, 2025-09-07:
Sebastien again wrote:

    C:\Users\fcgle\source\opencpn>git pull
    Enter passphrase for key 'C:\Users\fcgle.ssh\iD_ed25519':
    You are not currently on a branch.

This happens because you are in "detached HEAD" state. In git, HEAD is a pointer that tells git which commit you're currently working from. Normally, HEAD points to a branch name, and that branch points to a specific commit.

When you are on detached HEAD state, you're not "on" any branch. You're just sitting directly on a commit, and you cannot pull commits into it.

Here is how you can fix this:

# Create a new branch from where you are
git checkout -b timeline

# Set up tracking to the remote
git branch --set-upstream-to=sebastien/timeline

Now you should be able to pull

bdbcat, 2025-09-07:
plang...
re:
"It is not the single issue. OpenCPN is crashing while using the Intel/OpenGL driver. It cannot be a bug made by Intel as other OpenGL software do not crash on my PC. But it is said as in the current thread, that the problem is not in Opencpn... but in the Intel driver !'
I may beg to differ.  We have had demonstrated bugs, confirmed by many other GL developers and users, on Intel GL drivers.
OpenGL is a complex thing, and sometimes OpenCPN is "stretching" the drivers to their limits.  But I do not reject the idea of an OpenCPN bug completely.

Anyway, we can and should re-open an issue on this point (**without diverging into build problems)**, then we may make some progress this time.

closed 2025-09-08 by plang732

rgleason, 2025-09-09:
Yes, I found that the single batch file winconfig.bat  builds the world, and it works. I tried to follow the instructions carefully. There may be some situations where it isn't working but this is not one of them!

[Compiling OpenCPN with MSVC using repo opencpn-test.pdf](https://github.com/user-attachments/files/22239915/Compiling.OpenCPN.with.MSVC.using.repo.opencpn-test.pdf)

plang732, 2025-09-14:
> I agree there may be a bug in OpenCPN on Intel OpenGL for Windows. Especially late model Intel graphics hardware. I have an Intel GPU based machine that does not work with OpenCPN master. All other OpenGL apps work ok.
> 
> On my system, O does not crash but the chart canvas graphics frames and toolbar just stop updating shortly after O starts. Dashboard and other plugins seem to not be affected. About 1 out of 10 startups it will all work. So it may be some weird race condition that happens 90% of the time. I would like to find the cause but OpenGL debugging is a little out of my league.
> 
> I can create an issue for what I am seeing.

Hi,

You described the problem I had with the Intel driver. 
=> When the window is resize, minimized or maximized. 

So I set Windows to use the NVIDIA driver (it is OK)

**I found the cause of the problem yesterday in the OpenCPN sources. 
I fixed it on my build and tested it. For me, it is fixed...**

I suggested a solution to Dave. He is checking it.

My PC is new from January with The Intel Arc Graphics.

I don't think an new issue should be created.

Regards

plang732, 2025-09-14:
http://www.thoe.be/guest/opencpn.exe

rgleason, 2025-09-14:
Pierre,  I suppose you know about [GLview](https://www.realtech-vr.com/home/?page_id=142)?  It diagnoses boards for OpenGL

plang732, 2025-09-14:
> Pierre, I suppose you know about [GLview](https://www.realtech-vr.com/home/?page_id=142)? It diagnoses boards for OpenGL

Yes. It just checks the OpenGL hardware & driver. It does not help to analyze calling programs.

**I found the cause of the problem yesterday in the OpenCPN sources.
I fixed it on my build and tested it. For me, the issue with the Intel driver is fixed...
It is in Dave's hands now.**

There is another (not blocking) issue with the NVIDIA driver. 
At the moment, for me, it looks better to use the Intel driver than the NVIDIA one !

bdbcat, 2025-09-14:
Recommend that we move the OpenGL discussion elsewhere.
Not strictly related to this never-ending Windows build discussion.
Thanks

bdbcat, 2025-09-14:
Sorry, but I really feel the need to clarify on the above OpenGL discussion.
1.  I do not accept that there is a "problem with the OpenCPN sources"
2. The "workaround" that I am suggesting, and that Pierre has validated, involves adding a "glFinish()" call in certain situations.
3. So far, this workaround is only required for various intel embedded graphics chips.
4. A little research on glFinish() suggests that it should not be required for single-threaded OpenGL programs.  Further, there is a not-insignificant performance impact by using glFinish().  Indeed, OpenGL devs suggest strongly to not use glFinish() unless your have very specific graphics sync requirements in your design.
5. We had the glFinish() call in OpenCPN versions past, again only for intel drivers of a certain vintage.  We removed it for O5 in the understanding that intel had fixed their driver so that it would no longer be required.  Evidently this is not correct.

TDan:  You may like to experiment with this, adding glFInish() after the glSwapBuffers() call in glChartCanvas.cpp

plang732, 2025-09-15:
> Recommend that we move the OpenGL discussion elsewhere.
> Not strictly related to this never-ending Windows build discussion.
> Thanks

Hi Dave,

Please create a new issue for the OpenGL problem.
It will be necessary, for sure.

Regards, Pierre

plang732, 2025-09-15:
Hi

I created an OpenGL dedicated issue:
See https://github.com/OpenCPN/OpenCPN/issues/4773

Regards, Pierre