OpenCPN/OpenCPN #4757: Windows Compile fails

rgleason, 2025-09-09:
**Describe the bug**
Now wonder Pierre had great difficulty with the Windows Build setup!!!

Cloudsmith has tightened security and no longer permits unauthenticated downloads.

Cloudsmith has changed public access. Try downloading this!

https://dl.cloudsmith.io/public/alec-leamas/OCPNWindowsCoreBuildSupport.zip

We need to get this fixed.  @leamas  @transitterdan   Thaanks.
===========
You're absolutely right—Cloudsmith has tightened public access, and direct downloads like the one you linked now often require authentication or tokenized URLs. Here's how you can work around it:

### 🛠️ Steps to Download from Cloudsmith (When Public Access Is Restricted)

1. **Check the Project’s Documentation or GitHub**  
   The maintainer (in this case, likely Alec Leamas or the OpenCPN team) may have updated instructions or provided a new download method. Look for:
   - GitHub Releases
   - README or INSTALL files
   - Issue threads discussing Cloudsmith changes

2. **Use a Tokenized URL (if available)**  
   Some Cloudsmith repositories now require a signed URL or API token. If you’re part of the project or have contributor access:
   - Log into [[Cloudsmith](https://cloudsmith.io/)](https://cloudsmith.io)
   - Navigate to the repository
   - Generate a download token or find the signed URL under “Packages”

3. **Reach Out to the Maintainer or Community**  
   If you're blocked, consider posting in:
   - OpenCPN forums
   - GitHub Issues for the relevant repo
   - Emailing the maintainer directly (if contact info is public)

4. **Use `curl` or `wget` with Authentication (if needed)**  
   If you do get a token or signed URL, you can download via command line:
   ```bash
   curl -O "https://dl.cloudsmith.io/public/.../OCPNWindowsCoreBuildSupport.zip?token=YOUR_TOKEN"
   ```

---

If you're integrating this into a workflow or want to make it teachable for others in your club, I can help you draft a protocol for authenticated downloads and version tracking. Want to build that out?

leamas, 2025-09-09:
Havn't we talked about pasting I generated stuff here? Please don't.  Back later

rgleason, 2025-09-09:
I've tried the newer version of this, with the single batch file that builds the entire world.

```
Quick start:

From the Windows Desktop open x86 Native Tools Command Prompt for VS 2022 (or 2019). Then enter these 4 commands:
 git clone https://github.com/opencpn/opencpn
 cd opencpn
 .\buildwin\winConfig.bat
 .\build\opencpn.sln
```

At the end of the first command I was asked to execute 

`.\buildwin\configdev.bat   msbuild /noLogo /m -p:Configuration=Debug;Platform=Win32 .\build\opencpn.sln`

and then to issue

`.\build\opencpn.sln`

MSVC opened automatically and I then selected  Build > Build Solution

and it built, ending with

    63>Project not selected to build for this solution configuration 
    64>------ Skipped Build: Project: INSTALL, Configuration: Debug Win32 ------
    64>Project not selected to build for this solution configuration 
    ========== Build: 49 succeeded, 0 failed, 2 up-to-date, 15 skipped ==========
    ========== Build completed at 1:09 PM and took 27.594 seconds ==========

Now still in MSVC  pick Debug  > Start Debugging

    Opencpn opens at the Initializing screen. 

This process does work for me. It has changed the way my wxWidgets was setup for building plugins, but I will change it back.

This is pretty amazing how it does everything! Good job guys.

[Compiling OpenCPN with MSVC using repo opencpn-test.pdf](https://github.com/user-attachments/files/22239888/Compiling.OpenCPN.with.MSVC.using.repo.opencpn-test.pdf)