OpenCPN/OpenCPN #4769: Bad draw of line objects and area objects' boundary

superat, 2025-09-15:
**Describe the bug**
Line objects of some S-57 cells are not drawn correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Use S-57 cells in attachment: C1210012 and C1612353
2. Computer with OS windows 11
2.1 Run OpenCPN 5.10.2-0+12e5712.

3. Embedded device with OS Debian 11 bullseye
3.1 Run Executable file compiled from GitHub source code
3.2. Execute ./opencpn -D
3.3. Execute ./opencpn -P

4. You can see in the two different cases above, all of the line objects are not correctly drawn, area objects' boundary are also badly displayed.

**

[Cells.zip](https://github.com/user-attachments/files/22328260/Cells.zip)

[Screenshots.zip](https://github.com/user-attachments/files/22328261/Screenshots.zip)

**
C1210012:
[image]

C1612353:
[image]

bdbcat, 2025-09-16:
Please show some screenshots from eLaneViewer, of the same areas, for comparison.

superat, 2025-09-16:
@bdbcat 

[image]
[image]

Hakansv, 2025-09-16:
Just out of curiosity: What if you change to Display Category: "All" in OCPN?

bc38, 2025-09-18:
> bc38... so from your findings the encoding was not correct S-57 and you did retouch them?

For some reason, the Chinese office isn't optimizing ENC geometry, including adding SCAMIN, which is mandatory for ENC production.
However, all these modifications didn't fix the display issue with linear objects and some contours.

superat, 2025-09-22:
@bc38 Source S-57 data deficiency is a problem.