OpenCPN/OpenCPN #4772: Improve Distance Units System for Better Clarity and Functionality

sebastien-rosset, 2025-09-15:
**Is your feature request related to a problem? Please describe.**

The current units system in Options>Settings>Display is ambiguous and doesn't adequately serve different types of distance measurements. There can be confusion about which unit setting applies to which measurements. The current "Distance" and "Depth" categories are too broad and don't clearly indicate their scope of application, leading to unclear applicability of unit settings.

**Describe the solution you'd like**

Restructure the Units section in Options>Settings>Display into logical, well-defined categories with clear tooltips and appropriate unit choices:

1. **Horizontal Distance**:
    - Tooltip: Distance between geographic positions, visibility range, radar range
    - Units: Nautical miles, statute miles, kilometers, meters, *yards* (new)
    - Note: This is the existing "Distance" field, but renamed to make it less ambiguous. Because for example "depth" is a "distance".
2. **Depth:**
    - Tooltip: Distance below a reference surface (sea level, vessel draft)
    - Units: Meters, feet, fathoms
3. **Vertical Height:**
    - Tooltip: Tide levels, wave heights, air gaps, mast clearance, elevations above reference datum
    - Units: Meters, feet, centimeters
4. **Line/Chain Length**
     - Tooltip: Anchor rode, mooring lines, cables
     - Units: Meters, feet, fathoms

Key Benefits:

1. Clear separation of concerns - each category serves specific measurement types
2. Appropriate unit restrictions - prevents nonsensical combinations (tide levels in fathoms)
3. Enhanced user experience - tooltips explain exactly what each setting controls
4. Future-proof design - accommodates different customization needs
5. Resolves tide units issue - tide levels would use "Vertical Height" settings

**Describe alternatives you've considered**

Status quo with tooltips: Keep current system but add explanatory tooltips
Pros: No structural changes needed
Cons: Doesn't solve fundamental categorization issues or missing units

**Additional context**

This enhancement originated from discussion in issue #4735 where tide units were displaying incorrectly. The conversation revealed broader issues with the current units system that affect multiple areas of OpenCPN.

TwoCanPlugIn, 2025-09-15:
A great discussion piece Sebastien.

You could introduce the concept of major & minor units (not sure if that is the right terminology though).

For distance I personally prefer Nautical Miles and if less than 1 Nm,  I prefer metres.

For depth I prefer metres, and if less than 1m still prefer metres (eg 0.5m)

I can understand for those preferring imperial units, they may prefer for distance, Nautical Miles and Yards (or god forbid cables or chains!) and for depth fathoms and feet.

For boat and wind speed I always prefer knots, however many metric users would prefer wind speed as metres/ second.  

I agree that there does need to be specific units to distinguish depth units from distance units, and wind speed from boat speed. For vertical height (eg. Air Gap, GPS antenna height), I suggest a separate definition to that of depth.

rgleason, 2025-09-15:
I wonder if horizontal distance should also have feet (concept of major & minor units) for completeness.

sebastien-rosset, 2025-09-21:
Researching the standards seems to confirm that:

1. OpenCPN should have "height" category.
2. Tide level should be under the "height" category.

PiotrSol, 2025-09-21:
ECDIS uses MN/m units, there is no place for ft (and obviously no fathroom, yards, rods and other antique units) in profesional maritime. Apart of USA does any official chart provider still use ft?  Not many still provide paper charts anyway...   

AIS is metric, NMEA2k is metric. There's really no way back and we are discussing ädding yards.... what next? furlong? league? 
Or maybe cable length.... again different in US and the rest of the world!

rgleason, 2025-09-21:
1. OpenCPN is not ECDIS
2. It is a Chart Plotter Navigation program started in the US.
3. OpenCPN has a dedicated mode to try to emulate the ECDIS standards (and it comes quite close and is accepted by some authorities)>
4. OpenCPN does not have to follow ECDIS, although it is reasonable to do so when practicable.

sebastien-rosset, 2025-09-21:
@rgleason , agreed. In this particular case, my understanding is that ECDIS can serve the needs of users throughout the world, allowing them to customize their display units. Ie any user can decide for themselves what the display units are. That way everyone can be happy. 

The ECDIS and NOAA categories seem useful.

rgleason, 2025-09-22:
A nautical mile is approximately 6076 feet, Navy Subs use 2000 yards for a nautical mile and converse in close quarters using yards from the periscope to the rest of the crew.

sebastien-rosset, 2025-09-22:
> A nautical mile is approximately 6076 feet, Navy Subs use 2000 yards for a nautical mile and converse in close quarters using yards from the periscope to the rest of the crew. If you want to drop it, fine.

ok, but why would submarines use OpenCPN? I just want to have a minimum bar that a decent number of people would really use it, not just a nice to have or a theoretical scenario.

rgleason, 2025-09-22:
Ok I think just drop it.

rgleason, 2025-09-22:
> Ok I think just drop it. You have feet.  

Actually Horizontal Distance does not have feet.
I will make an Issue for enhancement.

nohal, 2025-12-09:
No, because this is an issue, not a pull request