OpenCPN/OpenCPN #4783: Trying to hit the tide icon with the (right) mouse button becomes a game of patience (icon too small)

BerndCirotzki, 2025-09-21:
**Describe the bug**
When trying to open the tide Window by clicking on it. its not opening immediately.
The Tide Icon seems to be a little to small.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the chart 
2. Click on (right) on the tide icon

**Expected behavior**
open immediately. (A bigger icon) to meet the icon

**Desktop (please complete the following information if applicable):**
 - Windows 11
 - Openc0pn opencpn_5.13.0-0+c32f75765

rgleason, 2025-09-21:
@BerndCirotzki  What is the display size (resolution) and what scale are you using?

BerndCirotzki, 2025-09-21:
... Ok, Icon size could be changed, that's OK. But when clicking direct on it the Windows not opening.

BerndCirotzki, 2025-09-21:
... Ok, I think it's only somtimes. Could be better but it is working. No Bug.

closed 2025-09-21 by BerndCirotzki