OpenCPN/OpenCPN #4790: Add Feet or Yards for Horizontal distance.

rgleason, 2025-09-22:
**Is your feature request related to a problem? Please describe.**
No feet (or yards) for horizontal distance? 

**Describe the solution you'd like**
Ability to select "Feet" in the Options > Units drop down for Horizontal Distance.

For Harbors and anchorages

sebastien-rosset, 2025-09-22:
IMHO, it does not make sense to add "feet" to Options -> Units -> Distance, as this will set "feet" for the entire application, regardless of the magnitude of the distance values. This would mean that large values would be expressed in feet. E.g. 1000 nautical miles is 6,076,120 feet. Nobody wants to see this kind of value.

- The [FormatDistanceAdaptive](https://github.com/OpenCPN/OpenCPN/blob/43f975aa54931d1e7f9958eac5a1232c92a498b2/model/src/navutil_base.cpp#L357-L378) function formats a distance value into a user-friendly string with an adaptive number of decimal places and an appropriate unit based on the magnitude of the distance and the user's preferred unit system.
- If the converted distance is very small (< 0.1) and the unit is a large-scale one (kilometers, miles, or nautical miles), it switches to a smaller unit (meters or feet) for better readability.

My take is that consistently using `FormatDistanceAdaptive()` is the way to go. That function is currently used for route "total length", route leg, total track, segment length, great circle length, horizontal clearance, AIS range, CPA. I can see it's not used everywhere there is a concept of horizontal distance. If you open an issue, make sure to identify the specific part of the UI that does not switch units for small distances.

rgleason, 2025-09-22:
if (usrDistance < 0.1 &&
      (unit == DISTANCE_KM || unit == DISTANCE_MI || unit == DISTANCE_NMI)) {
    unit = (unit == DISTANCE_MI) ? DISTANCE_FT : DISTANCE_M;
    usrDistance = toUsrDistance(distance, unit);
  }

What is the "?" above?

sebastien-rosset, 2025-09-22:
For small distances (< 0.1 in the main unit):
1.If user chose statute miles → switches to feet
2. If user chose kilometers OR nautical miles → switches to meters

This seems like a design oversight - US mariners who select nautical miles would probably expect small distances in feet, but the function gives them meters instead.

svMigration, 2025-11-18:
Thanks for this Rick. I've been asking for feet for the measure tool for a long time. I agree that yards are useless. And Nautical Miles / Feet is definitely what I would prefer. So tying Nautical Miles to meters is not a good idea. Thanks.
Bruce