OpenCPN/OpenCPN #4796: Exception  most recent Master

rgleason, 2025-09-25:
**Describe the bug**
Exception thrown at 0x6A7699A3 (wxmsw32u_core_vc14x.dll) in opencpn.exe: 0xC0000005: Access violation reading location 0x00000138.

C:\Users\fcgle\source\opencpn>git log --oneline -5
bfb84c7d4 (HEAD -> master, sebastien/master, sebastien/HEAD, origin/master, origin/HEAD) certificates.cpp: Remove old '#if 0'
84572c33e comm_drv_n2k_serial.cpp: Remove old '#if 0'
6eb784823 comm_drv_signalk_net.cpp: Remove old '#if 0'
1a070f80d mdns_service.cpp: Remove old '#if 0'
2a0e93c74 mdns_query.cpp: Remove old '#if 0'

---------------

The error is an access violation (0xC0000005) at the line `return true;` in `OCPNPlatform::BuildGLCaps`. This typically means a pointer is being dereferenced that is either invalid or points to memory that the process cannot access.

**Concise Error Description:**  
The exception is most likely caused by an invalid pointer dereference, specifically with the `pcaps` pointer (cast from `pbuf`) or possibly `gFrame` (used just before the exception). The immediate cause is not the `return true;` itself, but something that happened earlier in the function, corrupting memory or leaving a pointer in an invalid state.

**Why You Are Getting the Error:**  
The root cause is likely that the `pbuf` argument passed to `BuildGLCaps` is not a valid pointer to an `OCPN_GLCaps` object, or it has been deleted or corrupted before this function is called. Alternatively, `gFrame` may not be properly initialized, and calling `gFrame->Show()` could corrupt the stack or heap, leading to the access violation at the next statement. Since the value of `pbuf` is `0x04dfc0e8`, you should verify that this address is valid and that the object it points to is properly allocated and not deleted.

**Suggested Next Steps:**  
- Check where `GL_Caps` is allocated and ensure it is not deleted or overwritten before `BuildGLCaps` is called.
- Verify that `gFrame` is properly initialized before calling `gFrame->Show()`.
- Use memory diagnostic tools (like Valgrind or Visual Studio's debugger) to check for heap corruption or invalid pointer usage before this function is called.

If you can provide the definition of `OCPN_GLCaps` and the code where `GL_Caps` is allocated, I can help further pinpoint the issue.

**To Reproduce**
MSVS build and Debug  see screenshots.

[image]

[image]

[image]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS:  Master 5.13. most recent 
 -Win 11

rgleason, 2025-09-25:
I also got this notice on my remote repo for O

https://github.com/rgleason/OpenCPN/actions/runs/17992579498

sebastien-rosset, 2025-09-25:
I did this workaround in my local workspace:

```diff
diff --git a/gui/src/ocpn_app.cpp b/gui/src/ocpn_app.cpp
index 3745f03f3..8d5315b40 100644
--- a/gui/src/ocpn_app.cpp
+++ b/gui/src/ocpn_app.cpp
@@ -1224,20 +1224,17 @@ bool MyApp::OnInit() {
                        app_style, dockart);
   wxTheApp->SetTopWindow(gFrame);
 
+  // tell wxAuiManager to manage the frame BEFORE Initialize_3
+  // This prevents crashes when Initialize_3 triggers resize events
+  g_pauimgr->SetManagedWindow(gFrame);
+  g_pauimgr->SetDockSizeConstraint(.9, .9);
+
   //  Do those platform specific initialization things that need gFrame
   g_Platform->Initialize_3();
 
   //  Initialize the Plugin Manager
   g_pi_manager = new PlugInManager(gFrame);
 
-  // g_pauimgr = new wxAuiManager;
-  g_pauimgr->SetDockSizeConstraint(.9, .9);
-
-  // g_pauimgr->SetFlags(g_pauimgr->GetFlags() | wxAUI_MGR_LIVE_RESIZE);
-
-  // tell wxAuiManager to manage the frame
-  g_pauimgr->SetManagedWindow(gFrame);
-
   gFrame->CreateCanvasLayout();
 
   // gFrame->RequestNewMasterToolbar( true );
diff --git a/gui/src/ocpn_frame.cpp b/gui/src/ocpn_frame.cpp
index 331de92ab..7210ba202 100644
--- a/gui/src/ocpn_frame.cpp
+++ b/gui/src/ocpn_frame.cpp
@@ -2205,7 +2205,11 @@ void MyFrame::ODoSetSize() {
     GetPrimaryCanvas()->GetNotificationsList()->RecalculateSize();
   }
 
-  if (g_pauimgr) g_pauimgr->Update();
+  // Safety check: Only update AUI manager if it's properly initialized
+  // This prevents crashes during application startup when timing issues occur
+  if (g_pauimgr && g_pauimgr->GetManagedWindow()) {
+    g_pauimgr->Update();
+  }
 }
 
```

sebastien-rosset, 2025-09-25:
> Any clear idea why this change in startup order fixes the problem? That is to say, why does Initialize_3 generate resize events now, but not before?

One thing I could see is that when `OCPNPlatform::Initialize_3()` is called, the wxAuiManager ends up calling SetSizer, but  `g_pauimgr` may not yet been told which window to manage (via SetManagedWindow), so calling Update() will dereference a null pointer, leading to a crash.

The patch moves `g_pauimgr->SetManagedWindow(gFrame);` before `g_Platform->Initialize_3();`.
This ensures that if `Initialize_3()` (or anything it calls) triggers a resize event, the AUI manager is already set up and ready to handle it.

But this is a bit of guess work. I wish I could identify the specific commit with `git bissect` to be sure. \

```text
Thread 0 Crashed:
0 libwx_osx_cocoau_core-3.2.0.4.1.dylib 0x102ea8894 wxWindowBase::SetSizer(wxSizer*, bool) + 20
1 libwx_osx_cocoau_aui-3.2.0.4.1.dylib 0x1025c4ae0 wxAuiManager::Update() + 412
2 OpenCPN 0x100d9ea14 MyFrame::ODoSetSize() + 2856 (ocpn_frame.cpp:2208)
3 OpenCPN 0x100d8ecb4 MyFrame::OnSize(wxSizeEvent&) + 28 (ocpn_frame.cpp:2035)
4 libwx_baseu-3.2.0.4.1.dylib 0x1029b5554 wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) + 112
5 libwx_baseu-3.2.0.4.1.dylib 0x1029b541c wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 136
6 libwx_baseu-3.2.0.4.1.dylib 0x1029b6150 wxEvtHandler::TryHereOnly(wxEvent&) + 84
7 libwx_baseu-3.2.0.4.1.dylib 0x1029b6048 wxEvtHandler::ProcessEventLocally(wxEvent&) + 48
8 libwx_baseu-3.2.0.4.1.dylib 0x1029b5f94 wxEvtHandler::ProcessEvent(wxEvent&) + 112
9 libwx_baseu-3.2.0.4.1.dylib 0x1029b5e04 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
10 libwx_osx_cocoau_core-3.2.0.4.1.dylib 0x102ea69d4 wxWindowBase::SendSizeEvent(int) + 128
11 libwx_osx_cocoau_core-3.2.0.4.1.dylib 0x102d581d0 wxNonOwnedWindow::Show(bool) + 84
12 OpenCPN 0x100dcd4d0 OCPNPlatform::BuildGLCaps(void*) + 7664 (ocpn_platform.cpp:722)
13 OpenCPN 0x100dcaf50 OCPNPlatform::IsGLCapable() + 512 (ocpn_platform.cpp:834)
14 OpenCPN 0x100dcabc8 OCPNPlatform::Initialize_3() + 24 (ocpn_platform.cpp:517)     # Call to Initialize_3
15 OpenCPN 0x100d5db94 MyApp::OnInit() + 23296 (ocpn_app.cpp:1228)
16 libwx_osx_cocoau_core-3.2.0.4.1.dylib 0x102d540c8 wxApp::CallOnInit() + 152
17 libwx_baseu-3.2.0.4.1.dylib 0x10294f324 wxEntry(int&, wchar_t**) + 80
18 OpenCPN 0x100d55004 main + 40 (ocpn_app.cpp:338)
19 dyld 0x188e3dd54 start + 7184
```

leamas, 2025-09-25:
I can reproduce this on windows.

The failing commit is 7e541567af45  --  filenames: ocpCursor -> ocp_cursor (Google GL). Digging

leamas, 2025-09-25:
The difference is about the dockart provider is being handled as an argument to ocpn_frame instead of a global. I don't remember the exact details why I had to make this change; an educated guess is that it is about some chicken and egg dependency proiblem.

Following Sebastien's reasoning the following minimal patch solves problem for me:
```
$ git diff
diff --git a/gui/src/ocpn_app.cpp b/gui/src/ocpn_app.cpp
index 3745f03f3..b916da8e3 100644
--- a/gui/src/ocpn_app.cpp
+++ b/gui/src/ocpn_app.cpp
@@ -1224,8 +1224,6 @@ bool MyApp::OnInit() {
                        app_style, dockart);
   wxTheApp->SetTopWindow(gFrame);

-  //  Do those platform specific initialization things that need gFrame
-  g_Platform->Initialize_3();

   //  Initialize the Plugin Manager
   g_pi_manager = new PlugInManager(gFrame);
@@ -1238,6 +1236,9 @@ bool MyApp::OnInit() {
   // tell wxAuiManager to manage the frame
   g_pauimgr->SetManagedWindow(gFrame);

+  //  Do those platform specific initialization things that need gFrame
+  g_Platform->Initialize_3();
+
   gFrame->CreateCanvasLayout();

   // gFrame->RequestNewMasterToolbar( true );

```

leamas, 2025-09-25:
>  an educated guess is that it is about some chicken and egg dependency proiblem.

No. I just removed a global when there was an opportunity.  Removing globals is IMHO overall a Good Thing.

Hakansv, 2025-09-25:
Win11: The above patch in the "diff" solved it for me as well.

PR #4799 (2025-09-25): ocpn_app: Fix g_platform initialization order -- #4796
As discussed in bug, calling g_Platform->Initialize_3() before defining g_pauimgr's managed window might lead to a crash. Thus, move the call to a point where g_pauumgr is fully initialized-

Closes: #4796
commit 917efe97 (2025-09-25): ocpn_app: Fix g_platform initialization order - #4796

As discussed in bug, calling g_Platform->Initialize_3() before defining
g_pauimgr's managed window might lead to a crash. Thus, move the call
to a point where g_pauumgr is fully initialized-

Closes: #4796
closed 2025-09-25 by leamas