OpenCPN/OpenCPN #4801: Audio device checkbox without functionality

BerndCirotzki, 2025-09-25:
The checkbox has only one member ... and not realy and not really the possibility to select a device
So it makes sense to fill it with life.

For Windows I found this: https://github.com/f1nality/WindowsAudioOutput
maybe it could help. the wxsound, I think, does not have this option

It would be a good feature to select the Sounddevice.

**Describe alternatives you've considered**
deleting the checkbox from Optiondialog because it is superfluous. Is that an alternative?

[image]

leamas, 2025-09-25:
That checkbox is currently only meaningful for hosts using the PortAudio library i. e. Linux and Macos. The simplest solution is probably to use PortAudio  also on Windows; IIRC there is (was) a windows port available.  The current interface does not allow sound device enumeration.

OTOH, any windows user could redirect the sound to whatever device available, so  it's probably just not worth it.  An alternative solution could be to just disable this control on Windows with a suitable tooltip.

BerndCirotzki, 2025-09-25:
Of course, any Windows user can redirect the audio to a different device than the default. But that's not the point. The key is not to use the default device, as it's currently being used by another application. e.g., music output, amateur radio, etc.
It would be good to be able to use a separate USB speaker just for OpenCPN under Windows, too.
And if that works with PortAudio... what's the problem?

leamas, 2025-09-25:
> what's the problem?

The problem is to find someone with skills, time and motivation to actually make a PR. The devil is in the details, and until there is actual code this cannot really be evaluated.

BerndCirotzki, 2025-09-25:
Oops, I made this PR in 3.2023.

Maybe my programming skills weren't that good, so I'd dismissed it.
And now I hadn't thought this is a know-how issue on OpenCPN developers.

But apparently it seems I'm the only one who needs something like this. I don't know.

But really, displaying a checkbox that is supposed to do exactly that, but has no function, is pointless.

leamas, 2025-09-25:
> FFS, When is OpenCPN going to use native OS features when wxWidgets or other cross platform libraries are a poor substitute?

For Windows, as soon as someone steps up and codes a native backend. The structure is in place, we have already several backends including PortAudio, Android,  WxWidgets sound   and a generic command line one. Adding  a  native Windows backend should be straight forward.

Patches welcome :)

TwoCanPlugIn, 2025-09-28:
> Adding a native Windows backend should be straight forward

The actual code is relatively easy, more than enough examples on microsoft.com or for that matter chatgpt.

The difficult part for us less skilled programmers is plumbing it into OpenCPN, in this case into the sound libraries.

BerndCirotzki, 2025-09-29:
Okay, I've just been working with PortAudio and have come to a conclusion.
I have no problem with someone taking on this issue and finding a better solution. Let's go!
PortAudio PlaySound doesn't just only support the default device, otherwise I could have saved myself this.
Unfortunately, I have no idea about MswSound backend.

Added WinPortAudioSound.cpp
Removed all #ifdefs.