OpenCPN/OpenCPN #4804: Visualization issues, broken geometry in S-57

bc38, 2025-09-27:
This issue occurs in almost every S-57 chart. It manifests itself in the averaging of points in polygonal objects, causing gaps and sometimes fills that don't follow the polygon outline.
The examples include screenshots from CPN and ENC viewer, NOAA charts
win 11

[image]

[image]

[image]

[image]

bdbcat, 2025-09-28:
Could you give a geographic reference for the first image shown above, so that I may clearly reproduce the effect, and so that we can focus one one repeatable instance?

bdbcat, 2025-09-28:
I found the first point, Gladstone, MI.
And the second image, as well, please ?

bc38, 2025-09-28:
> Я нашёл первую точку — Гладстон, Мичиган. А второе изображение тоже, пожалуйста.

[image]

bdbcat, 2025-10-08:
We would like to see comparisons with previous builds, showing expected presentation and unexpected results in current build.