OpenCPN/OpenCPN #4807: MacOS:  Negative depth soundings unreadable on mac

rgleason, 2025-09-29:
**Describe the bug**
Reported by [roelanto](https://www.cruisersforum.com/forums/members/roelanto-254802.html)  on the Forum.
https://www.cruisersforum.com/forums/f134/negative-depth-soundings-unreadable-on-mac-296868.html

On my mac (opencpn 5.12.4), [depth](https://www.cruisersforum.com/forums/tags/depth.html) soundings on areas that dry out (so: negative depths) do not show well. See attached screenshot for a place where there is water (depth = 0.7) and where there is land (depth is -0.7). If you look very closely (very!), you see the intended underline for the depth -0.7.

Click image for larger version Name: Screenshot 2025-09-29 at 12.07.01.png Views: 6 Size: 20.9 KB ID: 305122

The font for depth soundings does not seem to be adjustable via 'Options' > 'User interface' > 'Fonts'.

Either I have a weird font (how do I change that?) or the default font on a mac is such that 'underline' doesn't work well.

What can I do?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. macOS]
 - Version [e.g. 22]

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: [e.g. Galaxy S20]
 - Andorid version: [e.g. 10]

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

Arne2024, 2025-10-03:
the same issue , negative soundings underline is too thin and can't be read. 
OpenCPN 5.12.2 , MacOS v15, Retina screen. 
see additional screenshot of ENC attached.

I would like to add that the font size of soundings cannot be increased,
it would be nice if possible to increase size.

Arne2024, 2025-12-27:
In current version 5.12.4 the ENC Sounding Font size can be controlled via the User Interface menu.
Use slider for ENC Soundings Factor to increase the font size. 
Negative soundings are recognised by the green background colour, so the underscore readability is not a major issue in my view.

I suggest this issue could be closed.

[image]

closed 2025-12-29 by rgleason